zephyrproject-rtos/zephyr #25452: Some USB samples targeting stm32 are malfunctioning

bpbradley, 2020-05-19:
# Description

This is a followup issue from #25293 

USB samples ```cdc_acm_composite``` and ```hid_cdc``` are compiling without warning or error, and flashing fine, but the device malfunctions when connected to a host (Unknown Device: Device Descriptor Request Failed on Windows 10). All other usb device functionality which I was able to test (cdc, hid, console) work correctly. The problematic samples work perfectly on nrf52840 dvk, as a control.

# Reproduce

1. ```west build -b nucleo_f429zi samples/subsys/usb/cdc_acm_composite```
2. ```west flash```
3. Connect device to host.
4. Wait for device to enumerate. Expected behavior is for two virtual com ports to appear, but instead the device failure occurs.

# Environment

- OS: Windows 10
- Toolchain: gnuarmemb
- Zephyr version: 2.3.0-rc1

# Additional Context

In an effort to eliminate bus frequency as a possible cause (which was a central problem in #25293) I checked usb dp/dm with an oscilloscope, and the frequency appears to be good (waveforms appear identical to nrf52840, as far as I can tell, but I could inspect more rigorously if needed).

jfischer-phytec-iot, 2020-05-19:
cdc_acm_composite has 6 endpoints + 1 control endpoint
hid_cdc has 8 endpoints + 1 control endpoint

STM32F429ZI does not have enough endpoints to work with this samples:
OTG_FS: 4 bidirectional endpoints
OTG_HS: 6 bidirectional endpoints

bpbradley, 2020-05-19:
> cdc_acm_composite needs 6 endpoints + 1 control endpoint
> hid_cdc needs 8 endpoints + 1 control endpoint
> 
> STM32F429ZI does not have enough endpoints to work with this samples:
> OTG_FS: 4 bidirectional endpoints
> OTG_HS: 6 bidirectional endpoints

I just discovered the same issue. I actually changed the number of bidirectional endpoints in the dts to 7, and it works. I am not entirely sure where the number 4 came from. The reference manual shows there is only one bidirectional, three in, and three out endpoints.

aurel32, 2020-05-19:
> So I am curious why the dts usb node isnt like this:
> 
> ```
> num-bidir-endpoints = <1>;
> num-in-endpoints = <3>;
> num-out-endpoints = <3>;
> ```
> 
> Currently it is like this
> 
> ```
> num-bidir-endpoints = <4>;
> ```

The explanation is given in this commit (or at least I hope): https://github.com/zephyrproject-rtos/zephyr/commit/7688f490652a58a4485d6eeb41aeeea15ba0fd77

bpbradley, 2020-05-19:
> > So I am curious why the dts usb node isnt like this:
> > ```
> > num-bidir-endpoints = <1>;
> > num-in-endpoints = <3>;
> > num-out-endpoints = <3>;
> > ```
> > 
> > 
> > Currently it is like this
> > ```
> > num-bidir-endpoints = <4>;
> > ```
> 
> The explanation is given in this commit (or at least I hope): [7688f49](https://github.com/zephyrproject-rtos/zephyr/commit/7688f490652a58a4485d6eeb41aeeea15ba0fd77)

Ah, I am glad you found that. That clears things up. Thanks!

carlescufi, 2020-05-20:
@erwango this seems resolved now, please close this if you think it's appropriate.

closed 2020-05-20 by erwango