zephyrproject-rtos/zephyr #25453: tests/posix/common fails on nucleo_wb55rg

ABOSTM, 2020-05-19:
**Describe the bug**
tests/posix/common fails on nucleo_wb55rg

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b nucleo_wb55rg tests/posix/common/
2. west flash
3. See error

**Expected behavior**
Test Passed

**Screenshots or console output**

> starting test - test_posix_realtime
> POSIX clock set APIs
> rts.tv_nsec= 833000000   mts.tv_nsec = 832000000
> 
>     Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/clock.c:81: test_posix_realtime: (rts.tv_nsec not equal to mts.tv_nsec)
> Nanoseconds not equal
> FAIL - test_posix_realtime

**Analysis**
The function ` test_posix_realtime()` read current time by 2 different ways and compare both results
```
	ret = clock_gettime(CLOCK_MONOTONIC, &mts);
	zassert_equal(ret, 0, "Fail to get monotonic clock");
	ret = clock_gettime(CLOCK_REALTIME, &rts);
	zassert_equal(ret, 0, "Fail to get realtime clock");
	zassert_equal(rts.tv_sec, mts.tv_sec, "Seconds not equal");
	zassert_equal(rts.tv_nsec, mts.tv_nsec, "Nanoseconds not equal");
```
Unfortunately, it appears that I face a specific timing where time is incremented by 1ms between the 1st read and the 2nd one, which causes `zassert_equal(rts.tv_nsec, mts.tv_nsec, "Nanoseconds not equal")` to fail.

pfalcon, 2020-05-20:
This is very similar to https://github.com/zephyrproject-rtos/zephyr/issues/25017. And this report doesn't include Zephyr git revision to reproduce against, so first suspicion one may get is that an outdated tree might have been used (before the fix for #25017).

ABOSTM, 2020-05-28:
@pfalcon , Sorry for the late answer.
Issue is reproducable on
sha1:3a2f8349fd75d3dcd4551e551848458ed5129737

Which means it includes fixes for #25017

ABOSTM, 2020-05-28:
Despite issue has been reproduced on STM32 boards, analysis show it is a generic issue (remove STM32 label )

ABOSTM, 2020-09-18:
Now that I better understand zephyr clock and TICK way of working, I proposed a new fix for this issue based on original k_usleep() synchro #28516.

So there is now 3 PR proposal for this issue:
* #25454 tests: posix: common: Rework comparison of REALTIME and MONOTONIC clock
* #28366 tests: posix: common: fix test for cross-clock equality
* #28516  tests: posix: common: k_usleep() synchro just before clock AP

@pfalcon @andyross @pabigot @d3zd3z 
Which would you like to keep ?

andyross, 2020-09-18:
I like #28516 personally, especially given that the tick alignment was already present and well-described in a comment.

FWIW: tick alignment is just plain required for tests that care about any kind of time measurement precision.  The choices are that, or to carefully calculate the appropriate slop for the given tested operations.

PR #28516 (2020-09-19): tests: posix: common: k_usleep() synchro just before clock API
tests: posix: common: k_usleep() synchro just before clock API

Move printk() out of k_usleep() tick alignment,
for a better tick synchro.

Fixes #25453
commit 2b445734 (2020-09-19): tests: posix: common: move printk() out of k_usleep() tick alignment

Move printk() out of k_usleep() tick alignment,
for a better tick synchro.

Fixes #25453
closed 2020-09-19 by nashif