zephyrproject-rtos/zephyr #25466: log build errors are not helpful

andrewboie, 2020-05-19:
This is from messing up one variadic argument to a LOG_ERR() statement:

```
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:507:25: note: in definition of macro 'UTIL_EVAL5'
  507 | #define UTIL_EVAL5(...) __VA_ARGS__
      |                         ^~~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:47: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:505:25: note: in expansion of macro 'UTIL_EVAL4'
  505 | #define UTIL_EVAL3(...) UTIL_EVAL4(UTIL_EVAL4(UTIL_EVAL4(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:25: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:47: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:505:36: note: in expansion of macro 'UTIL_EVAL4'
  505 | #define UTIL_EVAL3(...) UTIL_EVAL4(UTIL_EVAL4(UTIL_EVAL4(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:25: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:47: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:505:47: note: in expansion of macro 'UTIL_EVAL4'
  505 | #define UTIL_EVAL3(...) UTIL_EVAL4(UTIL_EVAL4(UTIL_EVAL4(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:504:25: note: in expansion of macro 'UTIL_EVAL3'
  504 | #define UTIL_EVAL2(...) UTIL_EVAL3(UTIL_EVAL3(UTIL_EVAL3(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:25: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:47: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:505:25: note: in expansion of macro 'UTIL_EVAL4'
  505 | #define UTIL_EVAL3(...) UTIL_EVAL4(UTIL_EVAL4(UTIL_EVAL4(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:25: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:47: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                                               ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:505:36: note: in expansion of macro 'UTIL_EVAL4'
  505 | #define UTIL_EVAL3(...) UTIL_EVAL4(UTIL_EVAL4(UTIL_EVAL4(__VA_ARGS__)))
      |                                    ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:25: note: in expansion of macro 'UTIL_EVAL5'
  506 | #define UTIL_EVAL4(...) UTIL_EVAL5(UTIL_EVAL5(UTIL_EVAL5(__VA_ARGS__)))
      |                         ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:506:36: note: in expansion of macro 'UTIL_EVAL5'
[... 1084 lines cut ...]
      |  ^~~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/sys/util.h:435:2: note: in expansion of macro 'Z_COND_CODE_0'
  435 |  Z_COND_CODE_0(_flag, _if_0_code, _else_code)
      |  ^~~~~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/logging/log_core.h:156:3: note: in expansion of macro 'COND_CODE_0'
  156 |   COND_CODE_0(NUM_VA_ARGS_LESS_1(__VA_ARGS__),\
      |   ^~~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/logging/log_core.h:158:11: note: in expansion of macro 'GET_ARGS_LESS_1'
  158 |        (, GET_ARGS_LESS_1(__VA_ARGS__))\
      |           ^~~~~~~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/logging/log_core.h:279:14: note: in expansion of macro 'Z_LOG_STR'
  279 |              Z_LOG_STR(__VA_ARGS__));\
      |              ^~~~~~~~~
/home/apboie/projects/zephyr/zephyr/include/logging/log_core.h:296:2: note: in expansion of macro '__LOG'
  296 |  __LOG(_level,           \
      |  ^~~~~
/home/apboie/projects/zephyr/zephyr/include/logging/log.h:40:25: note: in expansion of macro 'Z_LOG'
   40 | #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
      |                         ^~~~~
/home/apboie/projects/zephyr/zephyr/arch/arc/core/fatal.c:33:2: note: in expansion of macro 'LOG_ERR'
   33 |  LOG_ERR("lp_end: %08x lp_start: %08x lp_count %08x",
      |  ^~~~~~~
/home/apboie/projects/zephyr/zephyr/arch/arc/core/fatal.c:34:34: error: request for member 'lp_count' in something not a structure or union
   34 |   esf->lp_end, esf->lp_start, esf->lp_count);
```

andrewboie, 2020-05-19:
Oh lordy, this is even better:

```
/home/apboie/projects/zephyr/zephyr/arch/arc/core/fatal.c:34:10: warning: format '%x' expects argument of type 'unsigned int', but argument 4 has type 'log_arg_t *' {aka 'long unsigned int *'} [-Wformat=]
   34 |  LOG_ERR("r12: %08x  13: %08x  pc: %08x",
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   35 |   esf->r12, esf->r13, esf->r14, esf->r15);
      |                       ~~~~~~~~
      |                       |
      |                       log_arg_t * {aka long unsigned int *}
```

What is `log_arg_t *`?

Can we please log messages without having to go through the preprocessor Dance of A Thousand Veils? This is a horrifying end user experience. What is all this trying to accomplish??

nashif, 2020-07-25:
@nordic-krch  can you please take a look?

closed 2020-12-08 by github-actions[bot]