zephyrproject-rtos/zephyr #25468: FRDM_K82F DTS missing information for ADC-0

paulvohs, 2020-05-20:
To my understanding, in order to enable a CPU feature that is disabled by default, an entry of the form 
&feature {
    status="okay"
}; 
must be added into the board DTS. I added this to enable ADC functionality to the FRDM_K82F DTS and it failed to compile. 

A coworker of mine figured out that there is a missing
#io-channel-cells = <1>;
in the file zephyr/dts/arm/nxp/nxp_k8x.dtsi. I am not sure if this is a bug or intended behavior.

Steps to reproduce the behavior:
1. Add the following to zephyr/boards/arm/frdm_k82f/frdm_k82f.dts:
&adc0 {
    status = "okay";
};
2. Build any sample project with -b frdm_k82f.
3. See attached screenshot of error output.
[image]

I believe this is something that needed to be included in the FRDM_K82F DTS but was not, though I am unsure.

This is an important issue for anyone planning to utilize the ADC on the Freedom K82F board.

- OS is Mac
- Toolchain is Zephyr with Ninja
- Commit SHA is ed2d263e0c233df015448de8d45a0fae359afc8b

carlescufi, 2020-05-20:
@paulvohs enabling a feature is done through a combination of Devicetree and Kconfig. The particular error you are getting doesn't ring a bell but I am sure that @MaureenHelm or @galak will be able to point you to the right direction.

MaureenHelm, 2020-05-20:
Yes, this is a bug in the soc device tree. The dts fix is easy, but I'd like to also enable `tests/drivers/adc/adc_api` for this board while we're here. I can work up a PR, but I don't have access to the hardware right now so I'll need your help to test it.

PR #25502 (2020-05-21): dts: arm: Add missing adc property for kinetis k8x and kv5x socs
Adds the required #io-channel-cells property to kinetis k8x and kv5x
socs. We are now consistently defining this property for all
nxp,kinetis-adc16 compatible nodes.

Enables the adc instance and pinmux associated with arduino header pin
A2 on the frdm_k82 board. Adds adc to the board yaml to ensure we build
adc samples/tests for this board in CI.

Fixes #25468

@paulvohs I don't have a frdm_k82f board on hand to test this works at runtime, but I did check that it builds. Will you please try running `tests/drivers/adc/adc_api` on [... cut]
closed 2020-05-21 by carlescufi