zephyrproject-rtos/zephyr #25471: disco_l475_iot1 don't write last small block

nandojve, 2020-05-20:
**Describe the bug**
UpdateHub uses subsys/dfu/img_util structure to interact with system flash. The current subsys/dfu/img_util over disco_l475_iot1 isn't work when try right last block with small data block.

**To Reproduce**
1. clone https://github.com/UpdateHub/zephyr/commits/topic/uhu_wifi
2. set variables at samples/net/updatehub/prj.conf
CONFIG_UPDATEHUB_WIFI_SSID="wifi network"
CONFIG_UPDATEHUB_WIFI_PSK="password"
3. west build -b disco_l475_iot1 samples/net/updatehub
4. sign image
 ~/<zephyr_home_dir>/bootloader/mcuboot/scripts/imgtool.py sign --key ~/<zephyr_home_dir>/bootloader/mcuboot/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x6C000 ~/<zephyr_home_dir>/zephyr/build/zephyr/zephyr.hex ~/<zephyr_home_dir>/zephyr/build/zephyr/zephyr-1.0.0.hex
5. west flash --hex-file build/zephyr/zephyr-1.0.0.hex
6. See error on terminal

**Expected behavior**
Flash write without errors.

**Impact**
Can't use any update that works with subsys/dfu/img_util.

**Screenshots or console output**
```
uart:~$ *** Booting Zephyr OS build v2.3.0-rc1-138-g1d06829ef97b  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: none

*** Booting Zephyr OS build v2.3.0-rc1-179-g489f25467752  ***
[00:00:00.540,000] <inf> main: UpdateHub sample app started
[00:00:00.549,000] <inf> main: Confirming the boot image
[00:00:03.632,000] <inf> main: WIFI Connected
[00:00:03.660,000] <inf> main: Starting UpdateHub polling mode
[00:00:04.208,000] <inf> updatehub: No update available
[00:01:04.794,000] <inf> updatehub: Probe metadata received
[00:01:11.282,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000000, Size: 1024, Flush: 0
[00:01:11.849,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000400, Size: 1024, Flush: 0
[00:01:12.416,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000800, Size: 1024, Flush: 0
[00:01:12.983,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000c00, Size: 1024, Flush: 0
[00:01:13.550,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00001000, Size: 1024, Flush: 0
[00:01:14.117,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00001400, Size: 1024, Flush: 0
...
[00:03:42.028,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00042400, Size: 1024, Flush: 0
[00:03:42.557,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00042800, Size: 92, Flush: 1
[00:03:42.569,000] <err> flash_stm32: Write range invalid. Offset: 845824, len: 92
[00:03:42.579,000] <err> STREAM_FLASH: flash_write error -22 offset=0x000ce800
RC from stream_flash_buffered_write: -22
RC from flash_area_close: -22
[00:03:42.600,000] <err> updatehub: Error to write on the flash
[00:04:43.727,000] <inf> updatehub: No update available
uart:~$ 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.2
 - Commit ec9a413983

**Additional context**
At my environment is 100% reproducible

CC @erwango @ABOSTM @nvlsianpu @otavio

mtahirbutt, 2020-05-21:
it is a flash driver issue from zephyr-2.3-rc1. As STM32 flash drivers requires to write data in multiples of 8 which has ben changed somehow with zephyr-2.3-rc1.

erwango, 2020-05-22:
> it is a flash driver issue from zephyr-2.3-rc1. As STM32 flash drivers requires to write data in multiples of 8 which has ben changed somehow with zephyr-2.3-rc1.

@mtahirbutt write-block-size on L4 is 8 bytes indeed, but this is the way the HW IP works. So I'm not sure what SW change could have had an impact on this. Can you point a version where this used to work ?

mtahirbutt, 2020-05-22:
@erwango I was working earlier with 2.2.0 and it was fine. I do not remember exactly about 2.2.99, but the changes were in flash_synch() within subsys/dfu/img_util/flash_image.c. There was wrap around for smaller data chunks.
	if (ctx->buf_bytes < CONFIG_IMG_BLOCK_BUF_SIZE) {
		(void)memset(ctx->buf + ctx->buf_bytes, 0xFF,
			     CONFIG_IMG_BLOCK_BUF_SIZE - ctx->buf_bytes);
	}
Now I handle that in updatehub app.

There is some change with 2.3-rc1 that stream_flash is introduced empty_byte stuffing is no more there.

erwango, 2020-05-22:
If this is a consequence from a change on ` subsys/dfu/img_util/flash_image.c`, then @nvlsianpu might have a better understanding.

carlescufi, 2020-05-23:
@nvlsianpu and @de-nordic this looks like a regression.

nvlsianpu, 2020-05-23:
Will investigate on Monday.

PR #25584 (2020-05-26): storage/stream: fix possible unaligned write on buffer flush request
On buffer flush request it is very probably that write buffer
contains amount of data which is non write-block-size aligned.
Flash memory need to be write at minimal by write-block-size chunks.
This patch addresses mechanism which ensure such behavior by adding
missing bytes.

fixes #25471
commit 06827ecd (2020-05-26): storage/stream: fix possible unaligned write on buffer flush

On buffer flush request it is very probably that write buffer
contains amount of data which is non write-block-size aligned.
Flash memory need to be write at minimal by write-block-size chunks.
This patch addresses mechanism which ensure such behavior by adding
missing bytes.

fixes #25471

streamer buffer size should be multiple write-block-size of
the flash device in order to avoid unaligned flash write
request.
closed 2020-05-26 by carlescufi