zephyrproject-rtos/zephyr #25474: ipv6 client-server between ble's failed

Navin-Sankar, 2020-05-20:
**Describe the bug**
Trying to establish ipv6 client-server between two ble board's (disco_l475_iot1 as ipv6 server) , (96b_nitrgen as ipv6 client).  But getting bind failure error. Error no: 49

Source code for both the server and client is attached.

**Server Implementation**

**prj.conf**
````
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_SMP=y
CONFIG_BT_L2CAP_DYNAMIC_CHANNEL=y
CONFIG_BT_DEVICE_NAME="IPSP Server"
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_NET_L2_BT=y
CONFIG_INIT_STACKS=y
CONFIG_NET_PKT_RX_COUNT=10
CONFIG_NET_PKT_TX_COUNT=10
CONFIG_NET_BUF_RX_COUNT=20
CONFIG_NET_BUF_TX_COUNT=20
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=2
CONFIG_NET_MAX_CONTEXTS=6

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"

CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
````

````
#include <zephyr/types.h>
#include <stddef.h>
#include <errno.h>
#include <zephyr.h>
#include <sys/printk.h>

#include <bluetooth/bluetooth.h>
#include <bluetooth/hci.h>
#include <bluetooth/conn.h>
#include <bluetooth/uuid.h>
#include <bluetooth/gatt.h>
#include <sys/byteorder.h>

#include <stdio.h>
#include <net/socket.h>

#define PORT 2000
#define MESSAGE "Hello from server"

static void start_scan(void);
static struct bt_conn *default_conn;
struct k_sem sem;

static void device_found(const bt_addr_le_t *addr, s8_t rssi, u8_t type,
                         struct net_buf_simple *ad)
{
        char addr_str[BT_ADDR_LE_STR_LEN];
        int err;
[... 325 lines cut ...]
        if (sock < 0) {
                printk("Failed to create UDP socket %d \n", errno);
        } 

        ret = connect(sock, (struct sockaddr *)&addr6, sizeof(addr6));
        if (ret < 0) {
                printk("Cannot connect to UDP remote: %d \n", errno);
        }

        while(1) {
                ret = send(sock, MESSAGE, sizeof(MESSAGE), 0);
                if (ret < 0) {
                        printk("Send failed\n");
                }

                k_sleep(K_MSEC(2000));
        }
}
````

**Expected behavior**
BLE client want to connect to BLE server over ipsp.

**Impact**
But getting the bind failure error.

carlescufi, 2020-05-20:
@Vudentz would you mind taking a quick look at the code to see if you spot something?

Vudentz, 2020-05-26:
@jukkar I guess it failling in here:

```
 		if (!iface) {
 			NET_ERR("Cannot bind to %s",
 				log_strdup(net_sprint_ipv6_addr(
 						   &addr6->sin6_addr)));
 
 			return -EADDRNOTAVAIL;
 		}
```

Ive checked IPSP and it is doing calls to net_addr_pton and net_if_ipv6_addr_add, perhaps that is why bind works there and not in the given code or that is no longer required when setting CONFIG_NET_CONFIG_MY_IPV6_ADDR?

jukkar, 2020-05-27:
@Navin-Sankar could you add `CONFIG_NET_SHELL=y` and verify that your client BT network interface has `2001:db8::2` IP address (use `net iface` shell command for that).

Navin-Sankar, 2020-05-28:
@jukkar Interface is down.

Vudentz, 2020-06-01:
@jukkar Does the interface being down makes bind fails?

Navin-Sankar, 2020-06-02:
@jukkar enabled CONFIG_NET_SHELL=y ip address is only assigned when it connects to linux host.When I try to connect to zephyr ipsp server, ````net iface```` command shows interface is down and ip address is not assigned

Is there is any configuration to enable in server side?

jukkar, 2020-06-02:
> Does the interface being down makes bind fails?

That could be it.

> `net iface` command shows interface is down and ip address is not assigned

Especially if the network interface does not have yet IP address, then binding could fail.

carlescufi, 2020-07-09:
@Navin-Sankar is this still an issue or should we close it?

Navin-Sankar, 2020-07-10:
Still experimenting. I can't initiate the ble ipsp router up. Once the node device is paired with the central device, I issue the command **net iface**, it shows **Interface is down** in both device. Still I miss some net interface initialization in router side.

Navin-Sankar, 2020-09-11:
Didn't added the global ipv6 address(2001:db8::1) in both the ble ipsp router and ble ipsp node. That's the reason for connection failure.

closed 2020-09-11 by Navin-Sankar