zephyrproject-rtos/zephyr #25477: gic: interrupt priority and flags are swapped in device tree parsing

Abhishek-brcm, 2020-05-20:
I noticed in the `devicetree_unfixed.h` that Priority and Flags are swapped in the parsed `interrupts` property.

We have been specifying interrupts like below:
```
interrupts = <type irq_num irq_priority irq_flags>;
where,
type = GIC_SPI/GIC_PPI
irq_num = interrupt number relative to type base
irq_priority = IRQ_DEFAULT_PRIORITY(0xa) / desired priority
irq_flags = IRQ_TYPE_LEVEL/IRQ_TYPE_EDGE

```
For example:
```
&uart0 {
        interrupt-parent = <&gic>;
        interrupts = <GIC_SPI 24 IRQ_DEFAULT_PRIORITY
                      IRQ_TYPE_LEVEL>;
};
```
For this DT parsing script has generated below @ `devicetree_unfixed.h`:
```
#define DT_N_S_soc_S_uart_40020000_IRQ_IDX_0_VAL_irq 56
#define DT_N_S_soc_S_uart_40020000_IRQ_IDX_0_VAL_flags 10
#define DT_N_S_soc_S_uart_40020000_IRQ_IDX_0_VAL_priority 0
```
It can be seen that flags and priority has been swapped.

So, either we can modify parsing logic or swap DT properties.

stephanosio, 2020-05-20:
It's supposed to be: `interrupts = <type irq_num irq_flags irq_priority>;`

https://github.com/zephyrproject-rtos/zephyr/blob/fb35591f1eea49e97dabd2467d15df004690defa/dts/arm/xilinx/zynqmp.dtsi#L23-L31

stephanosio, 2020-05-20:
We need to fix the AArch64 dts.

Abhishek-brcm, 2020-05-20:
@stephanosio Thanks for the reference.
We have been following `qemu-virt-a53.dtsi` as example, which caused this error.

stephanosio, 2020-05-20:
> Should I go ahead and submit the correct version?

@Abhishek-brcm It would be great if you could do that.

Abhishek-brcm, 2020-05-20:
Sure, I will rectify arm64 DTs.

carlocaione, 2020-05-20:
While fixing this can we keep the Linux order by having the flag as latest parameter?

stephanosio, 2020-05-20:
> While fixing this can we keep the Linux order by having the flag as latest parameter?

@galak IIRC, you suggest the current order. What do you think?

Abhishek-brcm, 2020-05-20:
Holding off PR in case we decide to go other way- i.e. change order by having flags at last parameter.

@stephanosio @carlocaione Linux anyway does not support parsing interrupt priority, right?

carlocaione, 2020-05-20:
> Holding off PR in case we decide to go other way- i.e. change order by having flags at last parameter.
> 
> @stephanosio @carlocaione Linux anyway does not support parsing interrupt priority, right?

Correct.

stephanosio, 2020-05-20:
> @stephanosio @carlocaione Linux anyway does not support parsing interrupt priority, right?

For reference:
https://github.com/torvalds/linux/blob/115a54162a6c0d0ef2aef25ebd0b61fc5e179ebe/Documentation/devicetree/bindings/interrupt-controller/arm%2Cgic.yaml#L57-L79

galak, 2020-05-20:
I wanted to keep flags property in the same location as the linux binding.

galak, 2020-05-20:
Linux has the flags as the last field because it doesn't encode priority,  thus I wanted to maintain the order being the 3rd cell with linux.  So lets fix the DTS files to correctly encode this with flags being 3rd and priority being 4th.

Abhishek-brcm, 2020-05-21:
IMHO flags seems like "generic" name and should be put at last: <irq_type> <irq_num><irq_priority> <irq_flags>. However, I am fine with either way. @stephanosio @galak please let me know final decision..

stephanosio, 2020-05-21:
Taking into account that we are currently in the feature freeze/bug fix phase, it would be desirable to simply fix the problem and postpone any changes/enhancements for the future (i.e. make everything align with `dts/bindings/interrupt-controller/arm,gic.yaml`).

Abhishek-brcm, 2020-05-21:
okay, Have submitted change to make everything aligned w.r.t. existing GIC binding @ https://github.com/zephyrproject-rtos/zephyr/pull/25514

PR #25514 (2020-05-21): dts: arm: Fix incorrect interrupt cells order
In aarch64 DTs, priority and flags cells have been swapped,
fix the same.

Correct interrupt property per the GIC binding document
looks like:
interrupts = <irq_type irq_num irq_flags irq_priority>;

Fixes #25477
closed 2020-05-21 by carlescufi