zephyrproject-rtos/zephyr #25478: settings_runtime_set() not populating bt/cf

TTJO, 2020-05-20:
When calling settings_runtime_set() for bt/cf the function will return without errors, but when reading the characteristic the cf-value have not been updated.

I have debugged on this issue in a test setup with the aim to verify that changes to the CF characteristic is reported to the app and can be repopulated again after a reboot.

Steps to reproduce:
1. Connect Master
2. Read CF characteristic with test device (0)
3. Write to CF characteristic with test device (1)
4. Read CF characteristic again with test device - to verify update in stack (1)
5. Print CF value reported back to app (1)
6. Reboot Slave
7. Reconnect Master 
8. Print CF-value settings_runtime_set() is called with (1)
9. Print returncode of settings_runtime_set() (0)
10. Read CF characteristic with test device (0) <--- ERROR! Should be 1.

I have tried to add a lot of debug-prints around the call to settings_runtime_set() to make sure that the input is correct, and it is, but the CF-characteristic is just not updated.
I have not printed the address, but there are (and have ever only been) one Master connected, so the address-parsing would return with an error if it wasn't correct.

It is expected that the CF-characteristic is updated with the value of given in settings_runtime_set().

We are not dependent on CF yet, so it is not a showstopper, but it is definitely going to be needed later. 

The test runs in a Matlab RealTime setup with a Zephyr Slave and a Zephyr Master.

**Screenshots or console output**
app.xbleapp.bond_ctrl.bond_ctrl_data.peer_data.11.gatt_client_features is the value on the app-side.

```
[13:19:20] BleProxyZephyr.transfer(read/56:54:12:41:02:de/2b29)
cf_before_write_num = 0
[13:19:21] Reading attribute app.xbleapp.bond_ctrl.bond_ctrl_data.peer_data.11.gatt_client_features: HI-Left...(=0)OK
[13:19:21] Reading attribute app.xbleapp.bond_ctrl.save_counter: HI-Left...(=2)OK
[13:19:21] BleProxyZephyr.transfer(write/56:54:12:41:02:de/2b29/1)
[13:19:21] BleProxyZephyr.transfer(read/56:54:12:41:02:de/2b29)
cf_before_reboot_num = 1
[13:19:22] BleProxyZephyr.transfer(disconnect/56:54:12:41:02:de)
[13:19:22] BleProxyZephyr.transfer(is-connected/56:54:12:41:02:de)
[13:19:23] Checking condition...(1)...OK
[13:19:23] Reading attribute app.xbleapp.bond_ctrl.bond_ctrl_data.peer_data.11.gatt_client_features: HI-Left...(=1)OK
[13:19:23] Reading attribute app.xbleapp.bond_ctrl.save_counter: HI-Left...(=3)OK
[13:19:23] Reading attribute app.xbleapp.bond_ctrl.cf_debug_saves: HI-Left...(=1)OK
[13:19:23] Reading attribute app.xbleapp.bond_ctrl.last_save_cf_value: HI-Left...(=1)OK
[13:19:23] Reading attribute app.xbleapp.bond_ctrl.last_save_cf_value_len: HI-Left...(=1)OK
[13:19:23] Resetting device 'HI-Left' (dsp2)
[13:19:37] Reset successful for device ''
[13:19:37] Waiting for tag "app.xbleapp.es_app.es_app_env.state_bits.pairing_enabled_state" to be == the value of: 0  on connection HI-Left ...OK
[13:20:07] Reading attribute app.xbleapp.bond_ctrl.cf_debug_loads: HI-Left...(=1)OK
[13:20:07] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_peer: HI-Left...(=11)OK
[13:20:07] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_errors: HI-Left...(=0)OK
[13:20:07] BleProxyZephyr.transfer(connect/56:54:12:41:02:de)
[13:20:09] BleProxyZephyr.transfer(encrypt-connection/56:54:12:41:02:de)
[13:20:10] BleProxyZephyr.transfer(is-connected/56:54:12:41:02:de)
[13:20:11] Checking condition...(1)...OK
[13:20:11] BleProxyZephyr.transfer(getlog/56:54:12:41:02:de)
[13:20:11] PAUSE 15000 ms...OK
[13:20:26] BleProxyZephyr.transfer(read/56:54:12:41:02:de/2b29)
cf_after_reboot_num = 0
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.bond_ctrl_data.peer_data.11.gatt_client_features: HI-Left...(=1)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_loads: HI-Left...(=1)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_peer: HI-Left...(=11)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_last_cf_val: HI-Left...(=1)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_cf_count: HI-Left...(=1)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_load_errors: HI-Left...(=0)OK
[13:20:27] Reading attribute app.xbleapp.bond_ctrl.cf_debug_saves: HI-Left...(=0)OK
[13:20:27] PAUSE 1000 ms...OK
[13:20:28] Checking condition...(1 == 0)...FAILED
[13:20:28] Check has failed : 1 == 0

ans =

Error using CHECK (line 51)
Check has failed : 1 == 0

Error in BLE.bleBondCtrlSettingsLoadSave/Script (line 156)
                    CHECK('%s == %s', cf_before_reboot_num, cf_after_reboot_num);
```

fnde-ot, 2020-05-21:
Will try to reproduce with a standard settings backend to locate where the issue might be.

fnde-ot, 2020-05-22:
Added a PR that seems to fix this issue. Could this be backported?

FYI, while testing it seems that when the CF Robust Caching bit is set, the first GATT read after reboot/reconnect always returns BT_ATT_ERR_DB_OUT_OF_SYNC. Shouldn't we be able to at least read the db_hash without error?

PR #25543 (2020-05-23): bluetooth: host: fix wrong bt/cf settings loading
This commits fixes the loading of bt/cf settings into memory. Only data
was loaded and not the address.

Fixes: #25478.
Fixes: #25447
closed 2020-05-23 by carlescufi