zephyrproject-rtos/zephyr #25480: Unconditional source of shield configs can mess up configuration

pdunaj, 2020-05-20:
**Describe the bug**
Unconditional inclusion of shields kconfig is problematic as some of these duplicate a genuine config options and mess the zephyr configuration.
https://github.com/zephyrproject-rtos/zephyr/blob/master/Kconfig.zephyr#L23

see: `UART_INTERRUPT_DRIVEN` (uart option) and `boards/shields/sparkfun_sara_r4/Kconfig.defconfig`

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
Boards Kconfig files should be included when needed.

**Impact**
from annoyance to showstopper depending how much this mess can affect the particular project

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

erwango, 2020-05-20:
@pdunaj can you be more specific in your description ?

> Boards Kconfig files should be included when needed.

The line the line you're pointing at is not a board Kconfig file, but a shield Kconfig file and is already included under following conditions:
```
if SHIELD_SPARKFUN_SARA_R4
if NETWORKING
```

I don't exclude potential issues, but this deserve more detailed description.

pdunaj, 2020-05-21:
Hi @erwango , sure.

Below are three lines from Kconfig.zephyr
```
source "boards/shields/*/Kconfig.defconfig"                                                                              
source "$(BOARD_DIR)/Kconfig.defconfig"                                                                                  
source "$(SOC_DIR)/$(ARCH)/*/Kconfig.defconfig" 
```

These will include Kconfig from *one* board, *one* soc and *all* shields.
Once the Kconfig file is included it is adding options to the configuration regardless if this option is under `if` or has any `depends on`. Including all shield Kconfig files make all options that are defined there public for entire zephyr repo.
And unfortunatelly some of these define option `UART_INTERRUPT_DRIVEN ` which is already defined in uart driver. Now if you go to this option and see what selects it and what are the dependencies you get some strange staff about smarkfun sara instead of uart driver. But I guess this could as well be the other way round if inclusion order change.

Anyway - either shield should be included only when needed e.g.
```
if($SHIELD_DIR)
  source "$(SHIELD_DIR)/Kconfig.defconfig"
endif()
```
Or it should be make clear that when shield config file is defined it must never use the same config symbols as are already used in other places. (This actually should be enforced regardless as one will have problem if shield is used with conflicting driver).

erwango, 2020-05-22:
@pdunaj txs for the additional details.

> Once the Kconfig file is included it is adding options to the configuration regardless if this option is under 'if' or has any 'depends on'. Including all shield Kconfig files make all options that are defined there public for entire zephyr repo.

I'm surprised by this statement and will have more checks

> Anyway - either shield should be included only when needed e.g.
```if($SHIELD_DIR)
  source "$(SHIELD_DIR)/Kconfig.defconfig"
endif()```

This wold require something more elaborated as several shields could be supported at the same time, but this can be explored.

> Or it should be make clear that when shield config file is defined it must never use the same config symbols as are already used in other places. (This actually should be enforced regardless as one will have problem if shield is used with conflicting driver).

I don't think we can exlude all conflitcs between boards/shields and applications configurations.. At list since conflicts are possible between boards and applications already, shields are not the only trouble maker in this area.

Anyway, do you have any way to reproduce the issue you're mentioning? This would be easier to solve than theoretical discussions.

galak, 2020-05-26:
@pdunaj is there a simple build of hello world that can show the issue?  Would help with trying to debug or look at possible solutions.

pdunaj, 2020-05-27:
I am closing this issue. For some reason I assumed shield config is redefining the option but it is simply y selecting it, which is perfectly fine.

closed 2020-05-27 by pdunaj

Rob-Grady, 2025-10-03:
I'm not sure this issue is resolved.  I have the same problem.

`warning: UART_INTERRUPT_DRIVEN (defined at boards/shields\sparkfun_sara_r4\Kconfig.defconfig:24,
boards/shields\wnc_m14a2a\Kconfig.defconfig:17,
C:/ncs/v3.1.1/zephyr/soc/quicklogic/eos_s3/Kconfig.defconfig:14, drivers/serial/Kconfig:72) was
assigned the value 'n' but got the value 'y'. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_UART_INTERRUPT_DRIVEN and/or look up
UART_INTERRUPT_DRIVEN in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.`

I am getting warnings about kconfig conflicts with a shield I'm not even using. My project has:
CONFIG_UART_INTERRUPT_DRIVEN=n
 but this shield eos_s3 is changing to 'y'.

I don't use this shield.

erwango, 2025-10-06:
@Rob-Grady Please open a new issue to report the problem you're seeing.