zephyrproject-rtos/zephyr #25497: arbitrary futex pointers

andrewboie, 2020-05-20:
**Is your enhancement proposal related to a problem? Please describe.**
Right now, all k_futex objects, even though they can live in user memory, must be known to the kernel at build time.

This is because all k_futex pointers found are placed in the system's kernel object table generated by gperf. When a thread later makes a call like `k_futex_wait()`, the address of the futex is looked up in the table, and if found, the returned `struct z_object` has a link to that futex's associated `z_futex_data`, containing that futex's wait queue.

The current implementation was easy to do; we already had all the infrastructure for kernel objects and it was simple enough to add `k_futex` to the set of objects and generate its wait queue. But we can do better!

**Describe the solution you'd like**
We'd like user mode to call k_futex_wait() on *any* user-accessible memory location. This is how it works in Linux. We'd like to be able to do this without requiring heap allocations.

The basic idea is to maintain a table of active futexes, i.e. futexes that at least one thread is waiting on. This table is keyed by the futex address. When we go virtual memory layer, the table will need to be keyed by address plus whatever VM space the address is from.

On a `k_futex_wait()` syscall:
- Look up the futex address in the table. If present, wait on its wait_q_t.
- If not present, create a table entry for it, and then wait.

On a `k_futex_wake()` syscall:
- Look up the futex address in the table. If not found, nobody was waiting. Do nothing.
- Otherwise, wake one or wake all waiters. If this wakes everyone up, remove from the table.

The nice thing is that the maximum number of table entries needed is a function of the number of active threads in the system; worst-case is a scenario of all threads waiting on different futexes.

We don't have a general-purpose runtime hash table data structure in Zephyr for the table. But we can make a red-black tree, and that should work for lookups. We similarly use one already for dynamic kernel objects.

So I'm thinking we can extend z_futex_data to look like:

```
struct z_futex_data {
	_wait_q_t wait_q;
	struct k_spinlock lock;
	struct rbnode node;
}
```

And instantiate a `k_mem_slab` pool of these. The number of entries can either be the number of k_threads detected by the kobject mechanism, or a Kconfig. There would then be a `struct rbtree` to maintain the table which starts out empty.

**Describe alternatives you've considered**
None so far.

wentongwu, 2020-05-21:
if not so urgent, I will take this one.

andrewboie, 2020-05-21:
@wentongwu I think @jukkar was interested in picking this one up as it's needed for some network stack work he is doing, can you coordinate with him?

jukkar, 2020-05-22:
@wentongwu I do not mind you doing this work, if you have time. I was just volunteering for this as I would need this feature in net_mgmt API. I am not familiar with this part of kernel so it might take some time to get things working.

jukkar, 2021-03-06:
I do not have time to work on this, so unassigning.

closed 2026-09-05 by nashif