zephyrproject-rtos/zephyr #25499: Out of tree board: No sources given to target

biwecka, 2020-05-20:
Hi,

I'm trying to port the Arduino Nano 33 BLE Sense to Zephyr. I started working in the zephyr-tree of a west workspace by copying the board definition of the "nrf52840dk_nrf52840" and modified it. First I changed the definition of the LED pins and tested it with the sample/basic/blinky example. It worked as expected. But as I tried to to extract my board definition to be part of a seperate project (I think that's called a star topology with an application as the manifest repository) I ran into issues.
First my project structure:
```
west-workspace
 |
 +--- application
 |     +--- boards/arm/arduino_nano_33_ble_nrf52840
 |     +--- src/main.c
 |     +--- CMakeLists.txt
 |     +--- west.yml
 |     +...
 |
 +--- zephyr (got pulled in by 'west update')
 +--- ... (other zephyr dependencies like 'bootloader', 'tools', 'modules')
```

My `west.yml` contains
```yml
# Example T2 west.yml, using manifest imports.
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos
  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v2.2.0
      import: true
  self:
    path: application
```

My `CMakeLists.txt` contains
```cmake
cmake_minimum_required(VERSION 3.13.1)
# Re-direct the directory where the 'boards' directory is found from
# $ZEPHYR_BASE to this directory.
set(BOARD_ROOT ${CMAKE_CURRENT_LIST_DIR})

# This sample is only compatible with nrf52840_pca10065 because it has
# re-directed the 'boards' directory to a directory that only contains
# this board.
set(BOARD arduino_nano_33_ble_sense_nrf52840)

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(out_of_tree_board)

target_sources(app PRIVATE src/main.c)
```

For building my application I use this little bash script:
```bash
#!/bin/bash

export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk
source ../zephyr/zephyr-env.sh

rm -rf build
west build -c -p auto -b arduino_nano_33_ble_sense_nrf52840
```

The main.c in my "extracted" application directory contains exactly the same code as the blinky example.

When I execute my build script the following error occurs:
```
CMake Error at /[...]/zephyr/cmake/extensions.cmake:372 (add_library):
  No SOURCES given to target: drivers__serial
Call Stack (most recent call first):
  /[...]/zephyr/cmake/extensions.cmake:349 (zephyr_library_named)
  /[...]/zephyr/drivers/serial/CMakeLists.txt:3 (zephyr_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/[...]/application/build -S/[...]/application -GNinja -DBOARD=arduino_nano_33_ble_sense_nrf52840
```

As I'm relatively new to zephyr I'm not shure if I made a mistake or if this is a bug.
Either way I hope someone can help me with this problem.

WilliamGFish, 2020-06-30:
Hi,
I looked at this board pre-covid and the biggest issue to overcome is the boot-loader and lack of easily programming via a J-Link interface. There are pins on the bottom of the board but I've just not got around to soldering some pins on yet.

As it is a Arduino board it uses the BOSSAC programming tool which I haven't spent any time reviewing but there is support for within Zephyr. 

Maybe during July I'll get a another look.

tejlmand, 2020-06-30:
```
 CMake Error at /[...]/zephyr/cmake/extensions.cmake:372 (add_library):
  No SOURCES given to target: drivers__serial
Call Stack (most recent call first):
  /[...]/zephyr/cmake/extensions.cmake:349 (zephyr_library_named)
  /[...]/zephyr/drivers/serial/CMakeLists.txt:3 (zephyr_library)
```

This error happens because no files are added to drivers__serial library here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/CMakeLists.txt#L3-L33

as you are using a nRF52840 SoS, then you can use either uart or uarte.
You should have a line similar to this in your board dts file:
```
&uart0 {
	compatible = "nordic,nrf-uart";
```
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts#L143

this tells Zephyr that the uart is `nrf-uart` compatible, and thus the right driver will be included during build, and that library should not be empty.

closed 2020-08-06 by tejlmand