zephyrproject-rtos/zephyr #25507: tests/portability/cmsis_rtos_v2 failed on up_squared.

chen-png, 2020-05-21:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/portability/cmsis_rtos_v2
2. west flash ~/up_squared.sh
3. See error

**Screenshots or console output**

*** Booting Zephyr OS build v2.3.0-rc1-210-gac898bc49ea8  ***
Running test suite test_cmsis_v2_apis

starting test - test_kernel_apis
PASS - test_kernel_apis

starting test - test_delay
PASS - test_delay

starting test - test_thread_apis
PASS - test_thread_apis
...
starting test - test_mempool_dynamic
PASS - test_mempool_dynamic

**starting test - test_messageq
E: Page fault at address 0xd25ff8 (error code 0x3)
E: Access violation: supervisor thread not allowed to write**
E: PML4E: 0x0000000000d15827 Writable, User, Execute Enabled
E: PDPTE: 0x0000000000d14827 Writable, User, Execute Enabled
E:   PDE: 0x0000000000d0d023 Writable, Supervisor, Execute Enabled
E:   PTE: 0x8000000000d25001 Read-only, Supervisor, Execute Disable
E: RAX: 0x00000000001a4c04 RBX: 0x0000000000d1cd60 RCX: 0x000000000000000c RDX: 0x0000000000d26020
E: RSI: 0x0000000000d1cd60 RDI: 0x0000000000d1dbc8 RBP: 0x0000000000d26090 RSP: 0x0000000000d26000
E:  R8: 0x0000000000d26020  R9: 0x0000000000d4d280 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000d26020 R13: 0x0000000000d1dbc8 R14: 0x0000000000d1dbc8 R15: 0x0000000000000000
E: RSP: 0x0000000000d26000 RFLAGS: 0x0000000000010003 CS: 0x0018 CR3: 0x0000000000115000
E: RIP: 0x0000000000103f87
**E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x0000000000d18f90 (send_thread)**
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.2
 - Commit ID: ac898bc49

nashif, 2020-09-16:
get around the stack overflow with 
```
diff --git a/tests/portability/cmsis_rtos_v2/prj.conf b/tests/portability/cmsis_rtos_v2/prj.conf
index ccdf0628a9..bebe7530e0 100644
--- a/tests/portability/cmsis_rtos_v2/prj.conf
+++ b/tests/portability/cmsis_rtos_v2/prj.conf
@@ -14,3 +14,5 @@ CONFIG_CMSIS_V2_MEM_SLAB_MAX_DYNAMIC_SIZE=128
 CONFIG_CMSIS_V2_THREAD_MAX_COUNT=23
 CONFIG_CMSIS_V2_THREAD_DYNAMIC_MAX_COUNT=10
 CONFIG_TIMEOUT_64BIT=n
+CONFIG_CMSIS_V2_THREAD_MAX_STACK_SIZE=1024
+CONFIG_CMSIS_V2_THREAD_DYNAMIC_STACK_SIZE=1024
``
But then there is a test assert..

nashif, 2020-09-16:
```
START - test_thread_join
 - Creating thread B...
 - Creating thread A...
 * Thread B started.
 - Waiting for thread B to join...
 + Thread A started.
 * Thread B joining...
 - Thread B joined.
 + Thread A joining...
 - Waiting for thread A to join...

    Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/thread_apis.c:305: test_thread_join: (status not equal to osOK)
osThreadJoin thread A failed!
 FAIL - test_thread_join
```

aasthagr, 2020-10-14:
This bug is only partially resolved by the fix tests: cmsis_rtos_v2: Fix race condition #28490. There are still assertion errors to this testcases.

jenmwms, 2020-12-10:
Can you please help me understand why we are reporting this issue? I see that `up_squared` is excluded for this test case, so are we trying to resolve something in order to include it in testing? https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/portability/cmsis_rtos_v2/testcase.yaml

nashif, 2020-12-10:
> Can you please help me understand why we are reporting this issue?

the issue was reported before testcase.yaml was changed :-)

PR #28490 (2020-10-13): tests: cmsis_rtos_v2: Fix race condition
Both threads, 4 and the main thread (test_thread_join) where
joining thread5. If thread 4 is scheduled before the
main thread it will be dead before the main thread
had chance to join it. Adding a delay (following the test
pattern) to avoid this problem.

Fixes #25507
PR #30621 (2020-12-17): Fixes CMSIS v2 test on x86
* In test_event_flags_no_wait_timeout(), after creating thread1, there is a little delay to let it run to trigger FLAG1. However, in test_event_flags_signalled(), this is not being done, and on some platforms it triggers the assert in thread1() complaining the flag not being set. This adds the same delay in test_event_flags_signalled() and the this test passes for those previously failed platforms.
* Some of the thread tests are hard-coded with assumption that there is only one CPU. So limit the number of CPUs to 1 via kconfig.
* The conditions which prohibit running [... cut]
closed 2020-12-17 by jhedberg