zephyrproject-rtos/zephyr #25508: tests: power management

albertofloyd, 2020-05-21:
There are very few test cases that run power management though there are few boards that support the feature. There is a need to catch regressions and catch bugs when exercising power management when supported.
https://github.com/zephyrproject-rtos/zephyr/issues/20033 (generic)https://github.com/zephyrproject-rtos/zephyr/issues/23274 (SoC specific)

**Describe the solution you'd like**
A generic test case suite that cover the following features when SOC PWR MANAGEMENT is supported.
Tests should:
1) Perform operations that results in CPU to enter sleep state when supported in single thread or multithread case.
2) Uses hook to track entry/exit to sleep state and compare against a counter.
3) Has minimal benchmarks i.e sleep latency entry/exit

Have different KConfig configurations. i.e. logging enabled, logging disabled.

**Describe alternatives you've considered**

**Additional context**
Use board-specific sample code to perform checks above 
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/mec15xxevb_assy6853

s-kelvin, 2020-08-05:
Hi Alberto, I'm currently working on this issue. I went through the board-specific code. I wonder if I should only replace the lines where you directly assigned 0x240 or 0x10240 to the board's registers pins and then the case could be generic, since all the other codes seem already generic. If so, is it doable to use the common PM framework to set the board's sleep status instead of assigning values to the pin?

albertofloyd, 2020-08-05:
Yes, these gpio toggling can be removed or replaced with either logging messages or gpio led pins toggling (which I believe are defined across boards).  These just serve a debug purpose.

s-kelvin, 2020-08-13:
Every time the main is finished the program does not stop. Instead, it gives the error message below after four tests.
[00:01:32.385,864] <err> os: ***** USAGE FAULT *****
[00:01:32.385,894] <err> os:   Illegal use of the EPSR
[00:01:32.385,894] <err> os: r0/a1:  0x00000000  r1/a2:  0x0000000a  r2/a3:  0x00000000
[00:01:32.385,894] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000014
[00:01:32.385,894] <err> os:  xpsr:  0x60000000
[00:01:32.385,894] <err> os: Faulting instruction address (r15/pc): 0x00000014
[00:01:32.385,925] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:32.385,925] <err> os: Current thread: 0x00118590 (main)
[00:01:32.442,047] <err> os: Halting system

Have you encountered this before on your end? Any thought would be helpful!

s-kelvin, 2020-11-12:
Merged at [27733](https://github.com/zephyrproject-rtos/zephyr/pull/27733)

closed 2020-11-12 by s-kelvin