zephyrproject-rtos/zephyr #25509: OpenThread SED set link mode fail

ghlualven, 2020-05-21:
**Is your enhancement proposal related to a problem? Please describe.**
In openthread_start() the otThreadSetLinkMode() return fail (InvalidArgs) for SED. 
On my nRF52840dk_nRF52840, the initial mode is "rsdn", SED is only disable the "r", the result is "sdn", which is invalid for OpenThread.

openthread.c:
```
/* Sleepy End Device specific configuration. */
	if (IS_ENABLED(CONFIG_OPENTHREAD_MTD_SED)) {
		otLinkModeConfig ot_mode = otThreadGetLinkMode(ot_instance);

		/* A SED should always attach the network as a SED to indicate
		 * increased buffer requirement to a parent.
		 */
		ot_mode.mRxOnWhenIdle = false;

		otThreadSetLinkMode(ot_context->instance, ot_mode);
		otLinkSetPollPeriod(ot_context->instance, OT_POLL_PERIOD);
	}
```

mle_types.hpp:
```
   /**
     * This method indicates whether or not the device mode flags are valid.
     *
     * An FTD which is not rx-on-when-idle (is sleepy) is considered invalid.
     *
     * @returns TRUE if , FALSE otherwise.
     * @retval TRUE   If the device mode flags are valid.
     * @retval FALSE  If the device mode flags are not valid.
     *
     */
    bool IsValid(void) const { return !IsFullThreadDevice() || IsRxOnWhenIdle(); }
```

**Describe the solution you'd like**
Add additional statement for different device types:
For SED, the mode should be "s";
For MED, the mode should be "rs";
For FTD, the mode should be "rsdn";

rlubos, 2020-05-21:
It seems that you still have FTD configuration stored in your flash? That's why it starts as FTD, even though it's configured as MTD (SED to be specific). Try to erase your flash data area (`ot factoryreset` from the shell, or `nrfjprog -e` from your host) and see if the problem persists.

In general, FTDs and MTDs use different library variants so it's wise to erase the settings area before switching between device types.

carlescufi, 2020-05-21:
> Try to erase your flash data area (`ot factoryreset` from the shell, or `nrfjprog -e` from your host) and see if the problem persists.

If you use west to flash you can also use:
```
west flash --erase
```

ghlualven, 2020-05-22:
Confirmed "west flash --erase" can resolve the issue, thanks.

I wonder if it is worthy to improve the code to prevent this inconsistent occur?

rlubos, 2020-05-22:
I have no idea, for now, how could this be resolved in a clear way. I'm afraid the problems might show up not only with device mode stored in flash, but also other settings (for instance FTD will store information about connected children, this could be a problem for MTD as well). If you have a clean solution in your mind, feel free to propose.

ghlualven, 2020-05-23:
I have no idea for that, I am a newbie to OpenThread.

rlubos, 2020-05-25:
Let's close this issue then, IMO it's not really a bug.

closed 2020-05-25 by rlubos