zephyrproject-rtos/zephyr #25526: Sanity Check Fails:

mtahirbutt, 2020-05-21:
**Describe the bug**
A clear and concise description of what the bug is.

I wanted to contribute to the zephyr, therefore I ran sanity check on my local repo before creating PR. but it failed on my local version of repo.

What have you tried to diagnose or workaround this issue?

I tried to select specific board and tried again the sanity check but failed again.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; ./scripts/sanitycheck
2. cmake -DBOARD=any
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

The test should successfully run on the system.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I am not able to create PR because of this check.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Renaming output directory to /home/tahir/zephyr-latest2/https:/github.com/mtahirbutt/zephyr/zephyr/sanity-out.9
INFO    - JOBS: 1
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 4183 test configurations selected, 201981 configurations discarded due to filters.
INFO    - Adding tasks to the queue...

ERROR   - cc1352r1_launchxl         samples/boards/ti/cc13x2_cc26x2/system_off/sample.boards.ti.cc13x2_cc26x2.system_off FAILED: Build failure
ERROR   - see: /home/tahir/zephyr-latest2/https:/github.com/mtahirbutt/zephyr/zephyr/sanity-out/cc1352r1_launchxl/samples/boards/ti/cc13x2_cc26x2/system_off/sample.boards.ti.cc13x2_cc26x2.system_off/build.log
INFO    - Total complete:    1/4183   0%  skipped:    0, failed:    1
ERROR   - sensortile_box            samples/boards/sensortile_box/sample.board.sensortile_box.sensors FAILED: Build failure
ERROR   - see: /home/tahir/zephyr-latest2/https:/github.com/mtahirbutt/zephyr/zephyr/sanity-out/sensortile_box/samples/boards/sensortile_box/sample.board.sensortile_box.sensors/build.log
INFO    - Total complete:    2/4183   0%  skipped:    0, failed:    2
ERROR   - bbc_microbit              samples/boards/bbc_microbit/sound/sample.board.bbc_microbit.sound FAILED: Build failure

.................................................................................................................................

**Environment (please complete the following information):**
 - OS: (e.g. Ubuntu 18.04.4 LTS)
 - Toolchain (e.g Zephyr SDK 0.11.2)
 - Commit SHA: 251cb61e20e4a50116f03f82029413df7ce31d4e, zephyr-2.3-rc1

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-05-21:
> * mkdir build; ./scripts/sanitycheck
> * cmake -DBOARD=any
> * make
> * See error

Are you really running this? `any` is not a valid board

mtahirbutt, 2020-05-21:
> 
> 
> > * mkdir build; ./scripts/sanitycheck
> > * cmake -DBOARD=any
> > * make
> > * See error
> 
> Are you really running this? `any` is not a valid board

what i meant. I just used ./scripts/sanitycheck which considered all the boards during the check and created log there.

carlescufi, 2020-05-22:
@mtahirbutt which SHA of zephyr are you using? This builds fine locally for me:

```
$ west build -b cc1352r1_launchxl samples/boards/ti/cc13x2_cc26x2/system_off/
-- west build: generating a build system
-- Application: /home/carles/src/zephyr/zephyr/samples/boards/ti/cc13x2_cc26x2/system_off
-- Zephyr version: 2.3.0-rc1 (/home/carles/src/zephyr/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components:  Interpreter
-- Board: cc1352r1_launchxl
-- Found toolchain: zephyr (/home/carles/bin/zephyr-sdk)
-- Found west: /home/carles/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/carles/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/carles/src/zephyr/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl.dts
-- Generated zephyr.dts: /home/carles/src/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/carles/src/zephyr/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/carles/src/zephyr/zephyr/Kconfig
Loaded configuration '/home/carles/src/zephyr/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl_defconfig'
Merged configuration '/home/carles/src/zephyr/zephyr/samples/boards/ti/cc13x2_cc26x2/system_off/prj.conf'
Configuration saved to '/home/carles/src/zephyr/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/carles/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/carles/bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/carles/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/carles/src/zephyr/zephyr/build
-- west build: building application
[1/139] Preparing syscall dependency handling

[134/139] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       18208 B     360360 B      5.05%
      FLASH_CCFG:          88 B         88 B    100.00%
            SRAM:        5000 B        80 KB      6.10%
        IDT_LIST:         120 B         2 KB      5.86%
[139/139] Linking C executable zephyr/zephyr.elf
```

mtahirbutt, 2020-05-23:
I tried ./scripts/sanitycheck as mentioned in the zephyr website. but I stopped using that for now and will use online submission. if anything happes i will see this again. I used 251cb61, zephyr-2.3-rc1

carlescufi, 2020-05-25:
OK, closing this for now then

closed 2020-05-25 by carlescufi