zephyrproject-rtos/zephyr #25555: Unable to connect to Thread network (NRF52840DK)

lindemer, 2020-05-22:
**Describe the bug**
The `echo_client` sample fails to connect to my Thread border router when compiled with the `overlay-ot.conf` configuration file. I have uncommented the lines at the bottom of the file to enable joiner support. This appears to be a Zephyr issue; running `ot scan` from the shell shows that the network is up and is joinable, but `ot joiner start J01NME` always results in the error message `Join failed [NotFound]`. 

`ot masterkey` shows an exact match to the one set on the border router. I have tried running `sudo wpanctl commissioner add <eui64> 60 J01NME` on the border router which returns `Added Joiner <eui64>, timeout:60, PSKd:"J01NME"`. I'm not sure if this step is necessary, but it did not work in any case.

As a sanity check, I installed OpenThread on the same board following [this Google Codelabs tutorial](https://codelabs.developers.google.com/codelabs/openthread-hardware/) and am able to connect without issue. 

**To Reproduce**
Steps to reproduce the behavior:
1. `cd zephyr/samples/net/sockets/echo_client`
2. Uncomment the last lines of `overlay-ot.conf`
3. `west build -b nrf52840_pca10056 -t menuconfig -- -DCONF_FILE="prj.conf overlay-ot.conf"`
4. `west flash`

**Expected behavior**
I expected the board to join the network automatically, or at least connect after manually running the joiner commands from the shell.

**Impact**
I started working with BLE IPSP on Zephyr but switched to this approach after discovering a bug in the network stack (see [#25444](https://github.com/zephyrproject-rtos/zephyr/issues/25444)). I'm hoping one of these gets worked out soon so I can continue with my IoT project.

**Environment**
- OS: Ubuntu (Linux 5.3.0-51-generic) on VirtualBox
- Latest Zephyr SDK
- Zephyr v2.2
- Border router: Raspberry Pi 3B with Zolertia Firefly NCP, [certified releases](https://openthread.io/certification)

**Additional information**
I've noticed that the OpenThread configuration values set using KConfig don't seem to impact the values shown in the Zephyr shell. Is this a known issue?

retfie, 2020-05-23:
@lindemer try this changes into your overlay-ot.conf:

CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768
CONFIG_MBEDTLS_HEAP_SIZE=16384

This help me when I activate joiner support. It was related to OT_ERROR_NO_BUFS raised  by OpenThread stack during joining.

lindemer, 2020-05-24:
@retfie Thanks for the tip. I revisited the problem again today and the Zephyr board now seems to connect automatically with the unmodified `overlay-ot.conf`. I don't know what what caused the problem before, but I'll close this issue.

closed 2020-05-24 by lindemer