zephyrproject-rtos/zephyr #25569: Support for std::thread and std::this_thread

cfriedt, 2020-05-24:
**Is your enhancement proposal related to a problem? Please describe.**
It would be nice for C++ applications to be able to use [std::thread](https://en.cppreference.com/w/cpp/thread/thread) and [std::this_thread](http://www.cplusplus.com/reference/thread/this_thread/).

**Describe the solution you'd like**
To be able to use std::thread and this_thread. I can make a PR.

**Describe alternatives you've considered**
I recently created a PR to provide a sample for and to document the socketpair API (#25528) and was able to get threading working with a couple of defines and implementing [std::thread::join()](https://en.cppreference.com/w/cpp/thread/thread/join).

I also needed [nanosleep(2)](https://pubs.opengroup.org/onlinepubs/9699919799/functions/nanosleep.html) for [std::this_thread::sleep_for()](https://en.cppreference.com/w/cpp/thread/sleep_for)

However, there are also a couple of additional modes that will need testing. I thought it better to make a separate PR to address it.

**Additional context**

MrMarteng, 2024-06-29:
I'd love to have this functionality on Zephyr!

cfriedt, 2024-06-29:
@MrMarteng - we are one PR to zephyr-sdk away. The change that adds C++ threads support was unfortunately reverted as there were a couple of gaps in testing, but you can (IIRC) still download SDK release 0.16.6 which should have the required features.

See https://github.com/zephyrproject-rtos/sdk-ng/issues/751 for more info

vhouriecosta-sc, 2024-07-16:
I would also love to see this feature on Zephyr. Do we have any ETA of when will this feature be rolled out?

razvanphp, 2026-11-14:
In EV charging industry, the reference open-source (Linux) implementation EVerest uses `std::mutex` and there are efforts to port the code to MCUs. We succeeded on doing that on ESP-IDF, but it would be great to have the same on Zephyr-based MCUs, see https://github.com/EVerest/EVerest/issues/1851

Did things changed on this issue in the meantime? Can we support?