zephyrproject-rtos/zephyr #25592: Remove checks on board undocumented DT strings

erwango, 2020-05-25:
**Describe the bug**
Each time a new board is introduced, zephyrbot reports undocumented warnings for board's compatible:
```
-:358: WARNING:UNDOCUMENTED_DT_STRING: DT compatible string "olimex,stm32-p103" appears un-documented -- check ./dts/bindings/
#358: FILE: boards/arm/olimex_stm32_p103/olimex_stm32_p103.dts:13:
+	compatible = "olimex,stm32-p103", "st,stm32f103";

-:358: WARNING:UNDOCUMENTED_DT_STRING: DT compatible string "st,stm32f103" appears un-documented -- check ./dts/bindings/
#358: FILE: boards/arm/olimex_stm32_p103/olimex_stm32_p103.dts:13:
+	compatible = "olimex,stm32-p103", "st,stm32f103";
```

Since we're not requesting users to document these type of compatible, it would be nice not confusing contributors (that usually happen to be first time contributors) with these useless warnings.

**To Reproduce**
Steps to reproduce the behavior:
1. Push a new board port

See https://github.com/zephyrproject-rtos/zephyr/pull/25549#issuecomment-632670433

**Expected behavior**
Remove warnings if there is not action that could be done.

**Impact**
Useless churn

galak, 2020-09-08:
I think we should be asking people to add / update `vendor-prefixes.txt`

erwango, 2020-09-08:
One thing I can do is remove the soc compat ("st,stm32f103") as it seems it's only present on sSTM32 based boards.

Does CI check support `vendor-prefixes.txt` yet ?

galak, 2020-09-09:
> One thing I can do is remove the soc compat ("st,stm32f103") as it seems it's only present on sSTM32 based boards.
> 

Yeah, I think we've done that for other SoCs.

> Does CI check support `vendor-prefixes.txt` yet ?

What do you mean by that?

erwango, 2020-09-09:
> > One thing I can do is remove the soc compat ("st,stm32f103") as it seems it's only present on sSTM32 based boards.
> 
> Yeah, I think we've done that for other SoCs.

Ok, will do.
 
> > Does CI check support `vendor-prefixes.txt` yet ?
> 
> What do you mean by that?

If we add stuff such a file, will it prevent warnings to be issued in CI ?
I mean: today.

galak, 2020-09-10:
> > > Does CI check support `vendor-prefixes.txt` yet ?
> > 
> > 
> > What do you mean by that?
> 
> If we add stuff such a file, will it prevent warnings to be issued in CI ?
> I mean: today.

we already have the file, so I think if we add the board vendor prefixes it should make the warnings go away.

erwango, 2020-09-11:
> > > > Does CI check support `vendor-prefixes.txt` yet ?
> > > 
> > > 
> > > What do you mean by that?
> > 
> > 
> > If we add stuff such a file, will it prevent warnings to be issued in CI ?
> > I mean: today.
> 
> we already have the file, so I think if we add the board vendor prefixes it should make the warnings go away.

`olimex` and `st` are already present in our `vendor-prefixes.txt` file.
So, is that an issue in CI script?

closed 2020-09-30 by galak