zephyrproject-rtos/zephyr #25597: west sign fails to find header size or padding

paulgavrikov, 2020-05-25:
I am using west 0.7.1, Zephyr 2.2.0 and SDK 0.11.2 and would like to sign an image for mcuboot.
The following command works on my desktop, but fails on our jenkins CI: `west sign -t imgtool -- --key <key>.pem`

Error:
```
15:27:30 + west sign -t imgtool -- --key <key>.pem
15:27:31 === image configuration:
15:27:31 partition offset: 49152 (0xc000)
15:27:31 partition size: 421888 (0x67000)
15:27:31 text section offset: 0 (0x0)
15:27:31 === signed binaries:
15:27:31 bin: /var/lib/jenkins/workspace/build/zephyr/zephyr.signed.bin
15:27:31 Usage: imgtool sign [OPTIONS] INFILE OUTFILE
15:27:31 
15:27:31 Error: Invalid value for "-H" / "--header-size": Minimum value for -H/--header-size is 32
15:27:31 FATAL ERROR: command exited with status 2: /usr/local/bin/imgtool sign --version 0.0.0+0 --align 4 --header-size 0 --slot-size 421888 --key <key>.pem /var/lib/jenkins/workspace/build/zephyr/zephyr.bin /var/lib/jenkins/workspace/build/zephyr/zephyr.signed.bin
``` 

What could this be related to?

mbolivar-nordic, 2020-05-25:
The imgtool header size is the "text section offset", which is determined from the Kconfig symbol CONFIG_TEXT_SECTION_OFFSET. Indeed, 0 is less than 32.

I would check the CI build directory's zephyr/.config file and make sure the Kconfig symbol is defined correctly. If not, then there is a problem with the way the application is being built in CI.

paulgavrikov, 2020-05-25:
So, indeed the CI .config says CONFIG_TEXT_SECTION_OFFSET=0 whereas it's CONFIG_TEXT_SECTION_OFFSET=0x200 on my desktop. And I have found the issue. I forgot to commit the change that added the CONFIG_BOOTLOADER_MCUBOOT=y option ;)

closed 2020-05-25 by paulgavrikov

sw, 2021-01-13:
I ran into this problem as well with Zephyr 2.4.0

`TEXT_SECTION_OFFSET` has been renamed to `ROM_START_OFFSET`

The default value of 0x200 for that has been removed in commit 3b6642214721054a80b74e33cd86bd2335d7a68a