zephyrproject-rtos/zephyr #25600: wifi_eswifi: Unable to start TCP/UDP client

nandojve, 2020-05-26:
**Describe the bug**
After receive an **wifi_eswifi: Unable to start TCP/UDP client** the tcp/ip only works after restart the board.

**To Reproduce**
1. clone https://github.com/UpdateHub/zephyr/commits/topic/uhu_wifi
2. set variables at samples/net/updatehub/prj.conf
CONFIG_UPDATEHUB_WIFI_SSID="wifi network"
CONFIG_UPDATEHUB_WIFI_PSK="password"
3. west build -b disco_l475_iot1 samples/net/updatehub
4. sign image
 ~/<zephyr_home_dir>/bootloader/mcuboot/scripts/imgtool.py sign --key ~/<zephyr_home_dir>/bootloader/mcuboot/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x6C000 ~/<zephyr_home_dir>/zephyr/build/zephyr/zephyr.hex ~/<zephyr_home_dir>/zephyr/build/zephyr/zephyr-1.0.0.hex
5. west flash --hex-file build/zephyr/zephyr-1.0.0.hex
6. See error on terminal

**Expected behavior**
if for any reason devices have problems to create a TCP/IP client the future attempts must work.

**Impact**
Features stop to work.

**Screenshots or console output**
```
*** Booting Zephyr OS build v2.3.0-rc1-138-g1d06829ef97b  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: none

uart:~$ *** Booting Zephyr OS build v2.3.0-rc1-299-gd80ed410fe10  ***
[00:00:00.538,000] <inf> main: UpdateHub sample app started
[00:00:00.546,000] <inf> main: Confirming the boot image
[00:00:03.632,000] <inf> main: WIFI Connected
[00:00:03.657,000] <inf> main: Starting UpdateHub polling mode
[00:00:04.206,000] <inf> updatehub: No update available
uart:~$ updatehub run 
Starting UpdateHub run...
[00:00:20.094,000] <inf> updatehub: Probe metadata received
[00:00:26.580,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000000, Size: 1024, Flush: 0
[00:00:27.147,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000400, Size: 1024, Flush: 0
...
[00:01:03.463,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00010400, Size: 1024, Flush: 0
[00:01:04.030,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00010800, Size: 1024, Flush: 0
[00:01:04.292,000] <err> wifi_eswifi: Unable to start TCP/UDP client
[00:01:04.299,000] <err> updatehub: Cannot connect to UDP remote
[00:01:04.596,000] <err> updatehub: Could not receive data
Error installing update.
uart:~$ updatehub run 
Starting UpdateHub run...
[00:01:19.796,000] <err> wifi_eswifi: Unable to start TCP/UDP client
[00:01:19.803,000] <err> updatehub: Cannot connect to UDP remote
Invalid response
[00:02:04.365,000] <err> wifi_eswifi: Unable to start TCP/UDP client
[00:02:04.374,000] <err> updatehub: Cannot connect to UDP remote
[00:03:04.442,000] <err> wifi_eswifi: Unable to start TCP/UDP client
[00:03:04.451,000] <err> updatehub: Cannot connect to UDP remote
[00:04:04.462,000] <err> updatehub: Failed to create UDP socket
uart:~$ 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.2
 - Commit SHA 855b6c1

**Additional context**
This happen eventually.

nandojve, 2020-05-26:
CC @otavio @carlescufi @erwango @jukkar

erwango, 2020-05-26:
@nandojve if I'm correct, you have an esp shield, do you confirm this is working on the same board with esp shield (and eswifi deactivated ) ?
If yes, can you enable and provide WIFI logs ?

carlescufi, 2020-05-26:
See also https://github.com/zephyrproject-rtos/zephyr/issues/25621. CC @bwasim

bwasim, 2020-05-26:
In my case, Zephyr was acting as a TCP server and a client was able to successfully connect to it. However, the WiFi driver did not pass the client information to the application layer which caused NET stack to fail (whenever the application layer tried to send data to the client, was testing telnet with ES-WIFI)..

In this case, an AT command to the WiFi chip is failing. So this is different..

erwango, 2020-05-26:
@loicpoulain can  you have a check ?

nandojve, 2020-05-27:
> @nandojve if I'm correct, you have an esp shield, do you confirm this is working on the same board with esp shield (and eswifi deactivated ) ?
> If yes, can you enable and provide WIFI logs ?

Hi @erwango 

I noted that when use more buffers UpdateHub client works.
CONFIG_NET_PKT_RX_COUNT=8
CONFIG_NET_PKT_TX_COUNT=8
CONFIG_NET_BUF_RX_COUNT=20
CONFIG_NET_BUF_TX_COUNT=20

I think es-WIFI need more intensive tests because more buffers can delay the problem. ESP offload driver and frdm-k64 with ethernet are working fine using default buffer config.

As requested, below logs from ESP offload driver. I hope it can help to identify the problem.

```
*** Booting Zephyr OS build v2.3.0-rc1-138-g1d06829ef97b  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: test

[00:00:00.001,000] <inf> wifi_esp: Waiting for interface to come up
uart:~$ m[00:00:00.566,000] <inf> wifi_esp: ESP Wi-Fi ready
uart:~$ *** Booting Zephyr OS build v2.3.0-rc1-325-g36afe1cd643f  ***
[00:00:00.578,000] <inf> main: UpdateHub sample app started
[00:00:00.586,000] <inf> main: Confirming the boot image
[00:00:04.482,000] <inf> main: WIFI Connected
[00:00:04.503,000] <inf> main: Starting UpdateHub polling mode
[00:00:04.511,000] <dbg> wifi_esp_offload.esp_get: 
[00:00:04.519,000] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 3000
[00:00:04.528,000] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 192.168.10.199
[00:00:05.338,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.349,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.360,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:05.370,000] <dbg> wifi_esp_offload._sock_send: link 0, len 260
[00:00:05.458,000] <inf> updatehub: Probe metadata received
[00:00:05.467,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x0, data 0x0
[00:00:05.481,000] <dbg> wifi_esp_offload.esp_get: 
[00:00:05.489,000] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 3000
[00:00:05.498,000] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 192.168.10.199
[00:00:05.515,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.527,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.538,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:05.548,000] <dbg> wifi_esp_offload._sock_send: link 0, len 401
[00:00:05.608,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x0, data 0x0
[00:00:05.622,000] <dbg> wifi_esp_offload.esp_get: 
[00:00:05.629,000] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 3000
[00:00:05.639,000] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 192.168.10.199
[00:00:05.656,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.667,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:05.679,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:05.689,000] <dbg> wifi_esp_offload._sock_send: link 0, len 400
[00:00:05.749,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x0, data 0x0
[00:00:10.520,000] <dbg> wifi_esp_offload.esp_get: 
[00:00:10.528,000] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 3000
[00:00:10.537,000] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 192.168.10.199
[00:00:10.555,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:10.566,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:10.578,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:10.587,000] <dbg> wifi_esp_offload._sock_send: link 0, len 291
[00:00:10.739,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000000, Size: 1024, Flush: 0
[00:00:10.763,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:10.774,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:10.784,000] <dbg> wifi_esp_offload._sock_send: link 0, len 291
[00:00:10.927,000] <dbg> updatehub.install_update_cb: Flash: Address: 0x00000400, Size: 1024, Flush: 0
[00:00:10.951,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:10.962,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[00:00:10.972,000] <dbg> wifi_esp_offload._sock_send: link 0, len 291
[... 65 lines cut ...]
[00:00:59.752,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x0, data 0x0
[00:00:59.766,000] <dbg> wifi_esp_offload.esp_get: 
[00:00:59.773,000] <dbg> wifi_esp_offload.esp_connect: link 0, timeout 3000
[00:00:59.783,000] <dbg> wifi_esp_offload._sock_connect: link 0, ip_proto UDP, addr 192.168.10.199
[00:00:59.800,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:59.811,000] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x802f833, data 0x0
[00:00:59.823,000] <dbg> wifi_esp_offload.esp_sendto: link 0, timeout -1717986919
[... cut ...]
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: test

[00:00:00.001,000] <inf> wifi_esp: Waiting for interface to come up
uart:~$ m[00:00:00.567,000] <inf> wifi_esp: ESP Wi-Fi ready
uart:~$ *** Booting Zephyr OS build v2.3.0-rc1-325-g36afe1cd643f  ***
[00:00:00.579,000] <inf> main: UpdateHub sample app started
[00:00:00.587,000] <inf> main: Confirming the boot image
[00:00:04.487,000] <inf> main: WIFI Connected
[00:00:04.503,000] <inf> main: Starting UpdateHub polling mode
```

loicpoulain, 2020-07-07:
@nandojve could you please also increase debug log to DBG for eswifi.
As well, can you try to apply the patches from #26712

nandojve, 2020-07-11:
Hi @loicpoulain , I would like check but I'm stuck again with #25010.
CC @ABOSTM 

On Linux:
```
st-flash erase
west flash MCUBoot

*** Booting Zephyr OS build zephyr-v2.3.0-1003-g3e09400cc965  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.007,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: none
[00:00:00.013,000] <err> mcuboot: Unable to find bootable image
```
Than flash this new hello world
```
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..08ea5c1f9a 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,5 @@
 # nothing here
+CONFIG_BOOTLOADER_MCUBOOT=y
+CONFIG_IMG_MANAGER=y
+CONFIG_FLASH=y
+CONFIG_MPU_ALLOW_FLASH_WRITE=y
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 6c5c8a27dc..e5733d6f73 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -7,7 +7,20 @@
 #include <zephyr.h>
 #include <sys/printk.h>
 
+#include <dfu/mcuboot.h>
+
 void main(void)
 {
-       printk("Hello World! %s\n", CONFIG_BOARD);
+       int ret = -1;
+
+       printk("Hello World! 2 %s\n", CONFIG_BOARD);
+
+       /* The image of application needed be confirmed */
+       printk("Confirming the boot image\n");
+       ret = boot_write_img_confirmed();
+       if (ret < 0) {
+               printk("Error to confirm the image\n");
+       } else {
+               printk("Image is OK\n");
+       }
 }
```

```
*** Booting Zephyr OS build zephyr-v2.3.0-1003-g3e09400cc965  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.007,000] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.013,000] <inf> mcuboot: Swap type: none
*** Booting Zephyr OS build zephyr-v2.3.0-1003-g3e09400cc965  ***
Hello World! 2 disco_l475_iot1
Confirming the boot image
Error to confirm the image
```

Are there fuse to unlock all flash without any restriction?

erwango, 2020-09-16:
@nandojve would you mind rebasing https://github.com/UpdateHub/zephyr/commits/topic/uhu_wifi so we can have a test on recent  zephyr version ?

nandojve, 2020-09-30:
I'm wait an EVB board to isolate WiFi and create a Shield.
Removing Stale.

nandojve, 2020-11-02:
WIFI Shield and serial iface: #29507
UpdateHub uses CoAP over UDP. This issue can be related to #29538

erwango, 2020-11-02:
@nandojve now that there's a shield for eswifi, have you been able to reproduce this issue on other board ?

nandojve, 2020-11-02:
Hi @erwango , yes, I already tried UpdateHub and I got in trouble with frdm_k64f on early development using esWIFI with serial bus. Unfortunately I did not have enough time to look in deep but I've been suspecting that could be something related to UDP, after TagoIO's example start to work.
I have intention to test it again after #29507 and #29538 be merged.

closed 2021-01-16 by github-actions[bot]