zephyrproject-rtos/zephyr #25601: UART input does not work on mps2_an{385,521}

stephanosio, 2020-05-26:
**Describe the bug**
UART input does not work on `mps2_an385` and `mps2_an521` boards (in QEMU).

For example, the tests and samples that rely on the UART input functionality such as `drivers/uart/uart_basic_api` and shell are unable to accept any user input.

**To Reproduce**
In QEMU, run any tests and samples that require user input through the UART interface such as:

* drivers/uart/uart_basic_api
* samples/subsys/shell/shell_module

**Expected behavior**
UART user input is echo-ed on the input prompt.

**Impact**
Board usability is limited

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: 855b6c1a5ded091003d96a7b8d2b81662edbef70

**Additional context**
Broken UART driver suspected

carlescufi, 2020-05-26:
Removing the priority, we will go through it later today.

galak, 2020-05-26:
@microbuilder can you try one of these tests/samples on a real MPS2 to see if this shows up on real HW as well.

microbuilder, 2020-05-26:
Repeatable on actual HW as well (UART input also doesn't work, output from AN521 -> USB Serial does work fine, as is the case with QEMU).

PR #32335 (2021-02-15): drivers: uart: uart_cmsdk_apb: fix interrupt handling
The CMSDK uart interrupts for TX and RX can either be treated as a
signel interrupt line or distinct interrupts for TX & RX.  In the case
that they were distinct we didn't get the ifdef correct based on DTS.

If we have 2 interrupts in DTS we assume they are for TX & RX and thus
build the interrupt support for distinct TX & RX ISRs.

Also, cleanup handling of UART_2..UART_4 to be similar to how
UART_0/UART_1 code is using DT_INST_IRQN(x).

Fixes #30770
Fixes #25601
commit b063fe2a (2021-02-15): drivers: uart: uart_cmsdk_apb: fix interrupt handling

The CMSDK uart interrupts for TX and RX can either be treated as a
signel interrupt line or distinct interrupts for TX & RX.  In the case
that they were distinct we didn't get the ifdef correct based on DTS.

If we have 2 interrupts in DTS we assume they are for TX & RX and thus
build the interrupt support for distinct TX & RX ISRs.

Also, cleanup handling of UART_2..UART_4 to be similar to how
UART_0/UART_1 code is using DT_INST_IRQN(x).

Fixes #30770
Fixes #25601
commit ec0aa833 (2021-04-08): drivers: uart: uart_cmsdk_apb: fix interrupt handling

The CMSDK uart interrupts for TX and RX can either be treated as a
signel interrupt line or distinct interrupts for TX & RX.  In the case
that they were distinct we didn't get the ifdef correct based on DTS.

If we have 2 interrupts in DTS we assume they are for TX & RX and thus
build the interrupt support for distinct TX & RX ISRs.

Also, cleanup handling of UART_2..UART_4 to be similar to how
UART_0/UART_1 code is using DT_INST_IRQN(x).

Fixes #30770
Fixes #25601
closed 2021-02-15 by nashif