zephyrproject-rtos/zephyr #25604: USB: enable/disable a class driver at runtime

jli157, 2020-05-26:
**Is your feature request related to a problem? Please describe.**
In the current USB device stack, class drivers are statically selected during project configuration and can’t be changed at runtime. However, sometimes some class drivers could be required to be enabled or disabled at runtime. For example, the DFU driver could be disabled by default. And it will be helpful if it can be enabled by the application at runtime when the application wants to update the firmware. 

**Describe the solution you'd like**
All selected usb class drivers shall have a ‘enable(bool)’ api to enable / disable it from being discovered by the host at runtime when needed. And a setting value can be used to control if the class driver shall be initialized or disabled.

**Describe alternatives you've considered**
N/A

**Additional context**
Some system maintenance USB class drivers, such as CDC-ACM, DFU are not always needed but it is hoped to be included in the application firmware.

jfischer-phytec-iot, 2020-05-26:
@jli157 In fact, we have been discussing it for some time. Only more extensive than you described.

jli157, 2020-05-26:
> @jli157 In fact, we have been discussing it for some time. Only more extensive than you described.

Is there an existing issue covering the request? And what conclusions have you made?

jfischer-phytec-iot, 2020-05-26:
>Is there an existing issue covering the request? And what conclusions have you made?

No, still discussing. This is a major change, it affects class descriptor initialization and has to be implemented for all classes and applications in the tree. So that will take time.

emob-nordic, 2020-05-26:
There is no particular issue for that however there are few we would like to address during the rework
#20236 #19713 #24200

jfischer-phytec-iot, 2020-10-14:
Closing in favor of issue #29135

closed 2020-10-14 by jfischer-phytec-iot