zephyrproject-rtos/zephyr #25613: USB: CDC adds set line coding callback

xiongyihui, 2020-05-26:
**Is your feature request related to a problem? Please describe.**

Want to create a USB to UART application, but no API to find out when the baud rate is changed

**Describe the solution you'd like**
May provide an API to register callbacks similar to USB HID callbacks

**Describe alternatives you've considered**
Or add a weak function to [`cdc_acm_class_handle_req`](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/cdc_acm.c#L228), then we can override it.

jfischer-phytec-iot, 2020-05-26:
>Or add a weak function to cdc_acm_class_handle_req, then we can override it.

This approach is not acceptable and not necessary.
Please use uart_line_ctrl_get from UART API, see [samples/subsys/usb/cdc_acm](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/usb/cdc_acm/src/main.c#L121)

xiongyihui, 2020-05-26:
`uart_line_ctrl_get()` is OK if the USB CDC is opened once. But if the USB CDC is opened with a different baud rate at the second time, we don't know it.

jfischer-phytec-iot, 2020-05-26:
yes you have to poll it periodically.

xiongyihui, 2020-05-26:
Polling it periodically is not a good way. Maybe poll it when having new data to receive or to send.
But I still prefer an event-driven way to do that.

jfischer-phytec-iot, 2020-05-26:
> But I still prefer an event-driven way to do that.

Well the UART API does not provide that and usually baudrate does not change in the middle of a communication. To place an app hook in the processing of the control or class request or make it weak for a specific and not generic reason it not acceptable. Actually it is a bad API extension and will always make a problem during internal stack rework or redesign. It would also cause unnecessary latency and security issues. It is also not comparable wit HID because CDC ACM is self-contained implementation with clearly defined user interface (UART API).

xiongyihui, 2020-05-26:
OK. I'd try it  in my branch anyway.

closed 2020-05-26 by xiongyihui