zephyrproject-rtos/zephyr #25643: Replace or extend the existing crypto API to support secure elements and more algorithms

andysan, 2020-05-26:
**Is your enhancement proposal related to a problem? Please describe.**

Best-practice when deploying secure IoT devices is to use some sort of secure element to store cryptographic keys and other sensitive material. Such an element could be virtual using TF-M and PSA or using a hardware device such as the ATECC608 chip from Microchip.

When prototyping support for secure elements in Zephyr's LoRaWAN stack, I encountered serious limitations in the current API. Mainly: 1) Lack of support for MACs, 2) lack of support for key derivation, 3) lack of support for key management 

**Describe the solution you'd like**

I would like to see an API that supports the use cases above and the functionality provided by modern secure elements (including PSA) and hardware crypto accelerators. Specifically, the API needs to support opaque key management with (at least) the following operations:

- Key import for keys generated outside the secure element
- Key export for (e.g., for public keys or non-key data stored in a key slot)
- In-place key generation
- Key derivation using existing opaque keys and user-provided data

Supported algorithms need to include:
- AES
  - CBC
  - ECB
  - CTR
- CMAC (AES)

Nice to have features include:
- Common hash algorithms
- Public key crypto (e.g., ECC)

The API would need to support both hardware and software backends. Ideally, the software backends should be able to store keys using the settings subsystem to expose the same features as a "real" secure element.

**Describe alternatives you've considered**

As far as I can tell, there are three obvious candidates:

1. Stick with the current API and extend it
2. Switch to PKCS11
3. Switch to PSA

JSONROY, 2020-11-03:
Is there anyone doing this refactor? And is there any updates on this one?

rettichschnidi, 2024-02-26:
I need support for ECB. Will try to come up with PRs for tinycrypt, nRF52840 and SiM3U.

nashif, 2026-10-30:
Secure-element/algorithm needs addressed by PSA Crypto API via mbedTLS/TF-M; legacy include/zephyr/crypto API untouched. No activity since 2020.
include/zephyr/crypto/, modules/crypto/mbedtls

closed 2026-10-30 by nashif