zephyrproject-rtos/zephyr #25645: USB RNDIS driver can't work with Windows 10 (10.0.18363)

jli157, 2020-05-27:
**Describe the bug**
I tried to setup  the USB network between my NRF52840DK board and my windows 10 computer, by enabling `USB_DEVICE_NETWORK_RNDIS`. However, Windows 10 has problem to talk with my device blocked by the following two issues:
* Windows 10 can't install a proper RNDIS network driver for our device automatically. I have to manually choose "Microsoft Remote NDIS Compatible Device" driver. 

* can't set its IP address on its host side manually, as shown below:
[image]

**To Reproduce**
Steps to reproduce the behavior:
Enable `NETWORKING` and the following settings:

```
config USB_DEVICE_NETWORK_RNDIS
    default y
    select NET_IPV4
    select NET_TCP
    select NET_UDP
    select NET_SOCKETS
    select DNS_RESOLVER
    select SNTP
    select NET_CONFIG_SETTINGS
    select DNS_SERVER_IP_ADDRESSES
    select NET_SOCKETS_POSIX_NAMES
    select USB_DEVICE_OS_DESC

config NET_PKT_RX_COUNT
    default 4
config NET_PKT_TX_COUNT
    default 4
config NET_BUF_RX_COUNT
    default 12
config NET_BUF_TX_COUNT
    default 12
config DNS_SERVER1
    default "8.8.8.8"
config NET_CONFIG_MY_IPV4_ADDR
    default "192.168.33.111"
config NET_CONFIG_MY_IPV4_NETMASK
    default "255.255.255.0"
config NET_CONFIG_MY_IPV4_GW
    default "192.168.33.101"
config NET_CONFIG_PEER_IPV4_ADDR
    default "192.168.33.101"

```

**Expected behavior**
Window shall automatically recognize the USB RNDIS device and install the proper driver. And the host IP address on the USB network interface shall allow to be changed. 

**Impact**
USB Network on Windows doesn't work. 

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**

* [image]

* Zephyr v2.1

* NRF52840 DK board 

** Additional Context**
The same device works well with Linux (Ubuntu 18.01)

More logs are attached. 
[rndis_logs2.txt](https://github.com/zephyrproject-rtos/zephyr/files/4686317/rndis_logs2.txt)

jli157, 2020-05-27:
Verified the device works well with Windows 10.0.14393 Build 14393

jli157, 2020-05-27:
Actually, the IP address can  be set via the network adapter setting instead of "Ethernet" settings on the version 10.0.18363. 

However, the windows driver is still not able to be installed automatically.

jfischer-phytec-iot, 2020-05-27:
>     * Windows 10 can't install a proper RNDIS network driver for my device automatically. I have to manually choose "Microsoft Remote NDIS Compatible Device" driver.

> However, the windows driver is still not able to be installed automatically.

That is no reason to labels it as a bug here. Let's label it as question until one can describe/show why it is a bug in Zephyr's implementation.

jli157, 2020-05-27:
@jfischer-phytec-iot sorry for that! you are right. It is a question. Also, the new windows 10 could be more suspicious to make the driver not work.

carlescufi, 2020-06-10:
Closing this since this looks more like an issue on the Windows side. @jli157 please reopen if you find out this is indeed a problem on Zephyr's side.

closed 2020-06-10 by carlescufi