zephyrproject-rtos/zephyr #25652: smp_svr fails for nrf5340

hakonfam, 2020-05-27:
[placeholder for proper bug report when I have time]

Message is incorrectly split when transferring image in `smp_svr` for nRF5340

Perform FOTA for smp_svr using nRF Connect on iOS

The mcumgr response should be sent as a fragmented L2CAP-messag, instead it is split up.

See filenames in wireshark trace for relevant packets.

[trace.zip](https://github.com/zephyrproject-rtos/zephyr/files/4687442/trace.zip)

joerchan, 2020-06-08:
@hakonfam There is a possible fix here: https://github.com/zephyrproject-rtos/zephyr/pull/26018
Could you give it a try? Or update placeholder :)

hakonfam, 2020-06-09:
The fix did not affect this issue. Placeholder remains unchanged ;)

joerchan, 2020-06-24:
@hakonfam Will you have time soon to update the bug-report?

PR #26447 (2020-08-06): Bluetooth: host: Update configuration of L2CAP RX MTU setting
Remove dependency on CONFIG_BT_ACL_FLOW_CONTROL and use
CONFIG_BT_L2CAP_RX_MTU always to set L2CAP RX MTU.
The ATT MTU is set from two different KConfig options depending
on if CONFIG_BT_ACL_FLOW_CONTROL is enabled, which makes it
a confusing option and hard to provide a conf file that
supports multiple board configurations.

This changes the behavior when CONFIG_BT_ACL_FLOW_CONTROL and
CONFIG_BT_BUF_RX_LEN was used to set the L2CAP RX MTU, and by
extension the ATT MTU.

Fixes: #26536

Fix smp_svr Bluetooth configuration for when BT [... cut]
closed 2020-08-06 by carlescufi