zephyrproject-rtos/zephyr #25656: shields: Can't use multiple shields anymore

erwango, 2020-05-27:
**Describe the bug**
As described in https://docs.zephyrproject.org/latest/guides/porting/shields.html?highlight=shields#shield-activation, it used to be possible to use more than one shield in build command:
`west build -b None your_app -- -DSHIELD="x_nucleo_idb05a1 x_nucleo_iks01a1"`

This doesn't work anymore:
```
$ west build -b nucleo_f401re -- -DSHIELD="x_nucleo_idb05a1 x_nucleo_iks01a1"
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /local/mcu/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /local/mcu/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.3.0-rc1 (/local/mcu/zephyrproject/zephyr)
-- Board: nucleo_f401re
No shield named 'x_nucleo_idb05a1 x_nucleo_iks01a1' found
```

**Expected behavior**
Use of multiple shields should be possible

**Impact**
Block multiple shields application (eth + sensors, ...)

erwango, 2020-05-27:
@tejlmand, can you have a look ?

tejlmand, 2020-05-28:
Looking into this.
@erwango until the reason for this to not work when using space separated lists, you can use a proper CMake ;-separated list: 
```
west build -b nucleo_f401re -- -DSHIELD="x_nucleo_idb05a1;x_nucleo_iks01a1"
```
which will work.

erwango, 2020-05-28:
@tejlmand as this command is working, I'm totally fine to update the doc and get away with this.
I'm not particularly attached to the space separated list ....

tejlmand, 2020-05-28:
> @tejlmand as this command is working, I'm totally fine to update the doc and get away with this.
> I'm not particularly attached to the space separated list ....

As a long time CMake user, i'm familiar with ;-separated lists, but in Zephyr, there seem to be a tradition to also allow space-separated lists as arguments.

And because of that, and to stay consistent with `BOARD_ROOT`, `DTS_ROOT`, etc, I think we should still support space-separated lists.

I only made the comment as a work around (or the CMake way :wink: ), as I wasn't aware how easy it would be to fix.
But the issue is fixed here: #25716

PR #25716 (2020-05-29): cmake: resolved list issue when parsing shield list
Fixes: #25656

The commit a8e90cbfb30489c24139244dc3218c81c0950224 improved handling of
shields. Unfortunately it pops items from the SHIELD variable which
works well if SHIELD is a CMake ;-separated list, but if the list is
a space separated list, as may be provided by the user, then this
principle fails as reported in #25656.

This is fixed by keeping improved behaviour, while popping from an
alternative SHIELD-NOTFOUND list instead.
commit 8d3e822e (2020-05-29): cmake: resolved list issue when parsing shield list

Fixes: #25656

The commit a8e90cbfb30489c24139244dc3218c81c0950224 improved handling of
shields. Unfortunately it pops items from the SHIELD variable which
works well if SHIELD is a CMake ;-separated list, but if the list is
a space separated list, as may be provided by the user, then this
principle fails as reported in #25656.

This is fixed by keeping improved behaviour, while popping from an
alternative SHIELD-NOTFOUND list instead.
closed 2020-05-29 by carlescufi