zephyrproject-rtos/zephyr #25658: Some stm32 board do not run on current master (2.3 rc)

baw-serafin, 2020-05-27:
**Describe the bug**
Hello   World and blinky do compile and flash but the board never starts. No output / blinking LEDs

**To Reproduce**
Steps to reproduce the behaviour:
* go current master (I tested 6b54958a0e78e1de5d43910ebdc1ed281350586c)
* build for stm32g474re (only board I have here to test)
* run the samples

**Expected behavior**
Seeing something on console / Blinking LED

**Impact**
High I guess

**Environment (please complete the following information):**
 - OS: Debian
 - Toolchain Zephyr SDK 11.1
 - 6b54958a0e78e1de5d43910ebdc1ed281350586c

baw-serafin, 2020-05-27:
@erwango @FRASTM

erwango, 2020-05-27:
Provided a real board name. Given the tests we are running, I think this should not be extended to the whole STM32 board. I don't have this specific board but nucleo_g431 is running fine.

FRASTM, 2020-05-27:
Did you flash with openocd ?
west flash --openocd /usr/local/bin/openocd 
However flashing with pyocd makes the sample starts :
```
-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /home/r/zephyrproject/zephyr/build/zephyr/zephyr.hex
[====================] 100%
0003581:INFO:loader:Erased 14336 bytes (7 sectors), programmed 14336 bytes (14 pages), skipped 0 bytes (0 pages) at 14.39 kB/s

```

baw-serafin, 2020-05-27:
@FRASTM nope using the default west flash which is pyocd

erwango, 2020-05-27:
@FRASTM, can you confirm this is working on your side ?

gmarull, 2020-05-27:
Unfortunately I don't have this board to test.

FRASTM, 2020-05-27:
I can flash and run without pb, the samples/hello_world as well as samples/basic/blinky on my nucleo_g474re
(also  on nucleo_g431rb)

erwango, 2020-06-01:
@baw-serafin can you update SDK an have a new try?

erwango, 2020-06-04:
@baw-serafin any news on your issue?

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi