zephyrproject-rtos/zephyr #25659: ipv6 local link address support for applications

eyuval, 2020-05-27:
I am running zephyr on a system with multiple ethernet interfaces. I would like to be able to specify ipv6 link local addresses in applications, e.g. in the shell command 'net ping', but also in our tftp client application. Currently this doesn't seem to work. For this purpose it would be nice to be able to specify the network interface (e.g. in the same way as in linux) in the address itself and to be able to bind a socket to the interface or to indicate the interface in the sockaddr, proabably with the sin6_scope_id

jukkar, 2020-05-27:
>  I would like to be able to specify ipv6 link local addresses in applications, e.g. in the shell command 'net ping',

I discussed about this just recently in slack. We are just missing support for `-I <interface>` option in ping command. It should be quite easy to implement, patches are welcome if you want to implement one.

For the tftp one and other usecases, we could have the interface specified in scope_id. We are just starting to see devices with multiple network interfaces so support for this (using local addresses with multiple interface) is still not good. The system will select the interface correctly if global IP addresses are used but indeed there is some issues if you want to set the destination using local address. This is also most probably quite easy to implement. Currently I do not have any device with multiple interfaces so that I could test how much work this would require.

eyuval, 2020-05-27:
Thanks for your reply. Wouldn't it be better to encode the interface somehow in the address, e.g. as in linux with xxxxx%<interface identifier> notation, instead of adding an option in the ping command. In this way all applications would automatically benefit, when using sockaddr_in6 structures.

jukkar, 2020-05-27:
Well, ping in Linux has the `-I` option too.
About having generic parsing of xxxx% style IP strings, why not.

jukkar, 2020-05-29:
Just fyi, a PR for the -I option in ping was just submitted #25816

eyuval, 2020-05-29:
Great. Thanks for the information

carlescufi, 2023-11-15:
This is implemented now.

closed 2023-11-15 by carlescufi