zephyrproject-rtos/zephyr #25663: Documentation and features of IEEE802.15.4 drivers

markus-becker-tridonic-com, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

The documentation at https://docs.zephyrproject.org/latest is missing information about the available IEEE802.15.4 drivers, which features they support, and which platforms/boards they are supported on.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

kartben, 2023-09-19:
@fgrandel  fyi

fgrandel, 2024-07-24:
@nandojve Point 2 of the above list has been implemented by @rlubos in the meantime (thanks!). Are you still considering implementing (parts of) the Thread requirements in your driver? If so, this should be possible now.

fgrandel, 2024-07-24:
Marked as "good first contribution" as this is a rather simple documentation task that requires some initial invest to get going and can be taken forward from there by myself or driver maintainers.