zephyrproject-rtos/zephyr #25670: Possible Null pointer dereferences in /subsys/logging/log_msg.c

tluio, 2020-05-27:
I am not sure if (offset < available_len  && available_len <= LOG_MSG_HEXDUMP_BYTES_SINGLE_CHUNK && req_len > 0) can be true in log_msg_hexdump_data_op. If so, **cont** is NULL on line 452, 455 and 460.

nordic-krch, 2020-05-28:
This case will never happen. If `available_len <= LOG_MSG_HEXDUMP_BYTES_SINGLE_CHUNK` it means that message consist of single chunk and requested length + offset is trimmed to that. Then in `if (offset < chunk_len)` case is handling the case when data is read or write from first chunk. Since message is single chunk then `cpy_len` equals `req_len` and `req_len` will be zeroed in line 432 and while at 447 won't be executed.

tluio, 2020-05-28:
Ok. Thank you very much!

closed 2020-05-28 by tluio