zephyrproject-rtos/zephyr #25672: Bluetooth: Mesh: scan_start fails with synchronous bt_enable

trond-snekvik, 2020-05-27:
The Bluetooth Mesh fails in the `bt_le_scan_start` call in adv.c because of NRPA, but only if the Bluetooth subsystem is initialized without a `ready` callback. 

This happens because the scan_start call uses the hci_core `BT_ADV_ENABLED` flag to check whether the advertiser is in progress, while the controller uses the ull_adv state:
When initializing the mesh subsystem, the adv thread is started. It calls `bt_le_adv_start`, which calls `set_le_adv_enable_legacy`. The HCI command is processed by the controller, which sets the adv structure in ull_adv.c. Then, the HCI command response is enqueued in hci_core.c, and the adv thread goes to sleep. Next, settings are loaded, which starts the mesh subsystem and the scanner. When calling `bt_le_scan_start` from adv.c's `bt_mesh_scan_enable`, hci_core.c checks whether the advertiser is running, using its local `BT_ADV_ENABLED` flag. This flag won't be set until the adv_enable HCI command response is processed. hci_core decides that the advertiser isn't running, and sets the NRPA, which fails, because in the controller, the advertiser is already started.

I'm not sure why this works if bt_enable is called with a callback, but I assume it's a thread priority thing. It also works if I enable `CONFIG_BT_DEBUG_HCI_CORE`, which suggests it's a yield or race condition issue.

Bisecting the issue, it appears first in 915c510a0a, which looks like it is changing some scheduling aspects of the controller.

**To Reproduce**

Change the mesh sample to call `bt_ready` from main after the call to `bt_enable`, instead of as a callback:

```C
void main(void)
{
	int err;

	printk("Initializing...\n");

	/* Initialize the Bluetooth Subsystem */
	err = bt_enable(NULL);
	if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
		return;
	}

	bt_ready(0);
}
```

Then, provision and power cycle the device. It should immediately trigger the bug:

**Console log**

```
*** Booting Zephyr OS build v2.3.0-rc1-245-g451e559a22a4  ***
Initializing...
Bluetooth initialized
[00:00:00.008,819] <inf> fs_nvs: 2 Sectors of 4096 bytes
[00:00:00.Mesh initialized
008,819] <inf> fs_nvs: alloc wra: 0, f20
[00:00:00.008,819] <inf> fs_nvs: data wra: 0, 12c
[00:00:00.010,131] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.257,507] <wrn> bt_hci_core: opcode 0x2005 status 0x0c
[00:00:00.257,507] <err> bt_mesh_adv: starting scan failed (err -5)
```

trond-snekvik, 2020-05-27:
cc @jhedberg @carlescufi

jhedberg, 2020-05-28:
Didn't @joerchan add the `cmd_state_set` stuff to `hci_core.c` precisely because of this kind of stuff, i.e. so that we can set various flags through the RX priority thread?

cvinayak, 2020-05-28:
le_set_random_address failed, is the host trying to set the random address while having started the advertisement?

I will try to reproduce with hci core debug and check.

trond-snekvik, 2020-06-03:
> Is host calling hci_driver_send from multiple threads (controller's HCI interface is not designed to be reentrant)? and also, expects the controller to be blocking?

Yes, Mesh sends advertiser HCI commands from a separate thread, that's initialized like this: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/mesh/adv.c#L298-L301

> Host is initiating new HCI commands before previous command response is processed?

Yes, as a consequence of the separate thread.

Sounds like this is an issue hci_core should resolve then, for instance by mutexing the hci command/response handling when sent as a synchronous command. @joerchan

jhedberg, 2020-06-04:
@trond-snekvik that's not completely accurate. The `bt_hci_cmd_send()` API doesn't send anything but only queues the command to `bt_dev.cmd_tx_queue`. That queue is in turn processed by a single thread, `hci_tx_thread()`. Furthermore, this sending uses the `bt_dev.ncmd_sem` to control the sending, which in turn is a binary semaphore mapped to the `ncmd` parameter of Command Complete & Command Status events, so there should never be two pending commands at the same time, at least not from the controller's perspective.

trond-snekvik, 2020-06-04:
@jhedberg I see. The issue is still there though, since we're doing checks in the caller thread, which isn't subject to this semaphore. A bit odd that this started happening in a commit that only changed the controller, but I guess that just uncovered the already existing race condition.

carlescufi, 2020-09-10:
@joerchan you are still working on this right?

joerchan, 2020-09-14:
Yes.

The linked controller change is causing the TX thread to be unscheduled during the processing of the adv enable command.
So the main thread becomes active, and can continue to start the scanner, while the Mesh adv thread and TX thread is blocked. Since the command complete for adv enable has not been delivered then the current executing command in the TX thread is the adv enable command, but the host state of the advertiser is that it is not active.

The reason why changing to async is simply because this will executive in a cooperative thread and will continue to start the scanner after marking the Mesh advertiser thread as ready.

joerchan, 2020-09-15:
Update to bug description:

After commit c69a27b78970f3426 the behavior that triggered the bug has changed, and to get it back adding `CONFIG_SYSTEM_CLOCK_NO_WAIT=y` is needed.
Also on current master (36128bde19) the bug can sometimes happen on the sample with asynchronous bt_enable as well.

PR #28398 (2020-09-19): Bluetooth: host: Ignore failure to set passive scanner address
Ignore error when failing to set the passive scanner private address.
This can happen because in between the time we checked if the
advertiser was enabled and the time we execute the set random address
command the state of the advertiser could have changed in the
controller, and we only set the state once the command has completed
in the controller.

Fixes: #25672

Fix not setting NRPA before starting scanner.
Occurs with BT_PRIVACY=n and BT_EXT_ADV=y
commit f332d51d (2020-09-19): Bluetooth: host: Ignore failure to set passive scanner address

Ignore error when failing to set the passive scanner private address.
This can happen because in between the time we checked if the
advertiser was enabled and the time we execute the set random address
command the state of the advertiser could have changed in the
controller, and we only set the state once the command has completed
in the controller.

Fixes: #25672
closed 2020-09-19 by carlescufi