zephyrproject-rtos/zephyr #25682: Shell freezes with cout printf, prink on float

SocIt2Em, 2020-05-28:
**Describe the bug**
with shell enable. console appears frozen and shell prompt does not reappear

**To Reproduce**
double a = 2.333;
// any of the following will cause the shell prompt to not appear
std::cout << "cout float a is " << a << "\n";
printf("printf float a is %f \n", a);
printk("printk f float a is %f \n", a); // as expected. float not printed.

OUTPUT on uart
cout float a is 2.333
printf float a is 2.333000
printk f float a is %f
(NOTE: the shell prompt does not reappear)

LOG_X not tested.

**Expected behavior**
prints float number (not printk) AND Shell prompt returns.

**Impact**
Shell 

**Environment (please complete the following information):**
Zephyr kernel version 2.2.0
Version is 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
board: -  nrf52840_pca10056
cpluspluss is 201703
toolchain - gccarmnoneeabi

**Additional context**
proj.conf is: -
CONFIG_NEWLIB_LIBC=y  # so we can have math.h, etc...
#CONFIG_FLOAT=y  # for floating point? THIS CAUSES PROBLEMS and code will not compile
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y #for floating printf to work. include stdio.h then all good
CONFIG_CPLUSPLUS=y #should now be able to programm in c++
CONFIG_STD_CPP17=y # to say we are programming to this standard
CONFIG_LIB_CPLUSPLUS=y # Link with STD C++ library
CONFIG_HEAP_MEM_POOL_SIZE=1024
CONFIG_PRINTK=y  # should be able to printf floats
CONFIG_INIT_STACKS=y
CONFIG_THREAD_STACK_INFO=y
CONFIG_THREAD_MONITOR=y
CONFIG_BOOT_BANNER=n
CONFIG_THREAD_NAME=y
CONFIG_POSIX_CLOCK=y
CONFIG_UART_CONSOLE=y #on by default in nrf52840_pca10056_defconfig
CONFIG_STDOUT_CONSOLE=y

#shell
CONFIG_SHELL=y
CONFIG_SHELL_PROMPT_UART="zevice uart:~$ "
CONFIG_KERNEL_SHELL=y
CONFIG_DEVICE_SHELL=y
CONFIG_GPIO_SHELL=y
CONFIG_ADC_SHELL=y

#log
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_LOG_IMMEDIATE=y
CONFIG_ADC_LOG_LEVEL_INF=y
CONFIG_LOG_FUNC_NAME_PREFIX_ERR=y
CONFIG_LOG_FUNC_NAME_PREFIX_INF=y
CONFIG_LOG_FUNC_NAME_PREFIX_DBG=y
CONFIG_LOG_FUNC_NAME_PREFIX_WRN=y

#gpio
CONFIG_GPIO=y #on by default in nrf52840_pca10056_defconfig

#pwm
CONFIG_PWM=y
CONFIG_PWM_1=y
CONFIG_PWM_LOG_LEVEL_DBG=y

#adc
CONFIG_ADC=y
CONFIG_ADC_ASYNC=y
CONFIG_ADC_0=y

carlescufi, 2020-05-28:
@pabigot and @andrewboie any opinions on this one?

carlescufi, 2020-05-28:
@SocIt2Em would you mind trying with the latest master as well?

pabigot, 2020-05-28:
I can't verify this on current master using https://github.com/pabigot/zephyr/commits/reproduce/25682 which adds a shell to a C++ version of hello world with a command that emits floating point values.

If whatever is producing the output in your application is in a non-shell thread maybe that's the problem: I would not be surprised if having mutliple things try to control a single uart would cause problems.

nordic-krch, 2020-05-29:
shell uart backend may use interrupt API and there was #22355 merged after 2.2 which should fix concurrent use of polling and interrupt.

SocIt2Em, 2020-06-01:
@carlescufi currently I am using platformio which uses Zephyr version 2.202. Did an update command, but said 2.2 is uptodate. 
I suspect you are right @pabigot as I commented out all offending lines and compiled  and ran ok. however when I added another class instance, got the shell frozen problem again. 
Also I did not check LOG_X as for some reason I could not get it Logging working again. 
The only thread I am running is the main thread. 
Will give more feedback when getting my  ADC threads running and LOG_X.
p.s sorry for delay in responding. long weekend and we were enjoying our new freedoms

SocIt2Em, 2020-06-02:
in prj.conf I added
CONFIG_LOG_BACKEND_UART=y
Still can get shell to freeze for some unknown reason or for float, however I get this error back on uart:-

[00:00:00.052,886] <err> os.usage_fault: ***** USAGE FAULT *****
[00:00:00.059,906] <err> os.usage_fault:   Illegal load of EXC_RETURN into PC
[00:00:00.068,084] <err> os.esf_dump: r0/a1:  0x00000001  r1/a2:  0x000172f3  r2/a3:  0x80000000
[00:00:00.078,002] <err> os.esf_dump: r3/a4:  0x200045e4 r12/ip:  0x20004180 r14/lr:  0x00019f0b
[00:00:00.087,921] <err> os.esf_dump:  xpsr:  0x20004400
[00:00:00.094,207] <err> os.esf_dump: Faulting instruction address (r15/pc): 0xffffffed
[00:00:00.103,302] <err> os.z_fatal_error: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.127,868] <err> os.z_fatal_error: Current thread: 0x20000428 (shell_uart)
[00:00:00.151,611] <err> os.k_sys_fatal_error_handler: Halting system

Is that of any help?

pabigot, 2020-06-03:
@SocIt2Em To follow up: as far as I can tell this works in master.  We aren't going to be able to fix something if it's only present on previous releases.

We can't do much about platformio, but perhaps they'll update when 2.3.0 comes (should be no later than next week).  Please update this when you've tested it on a more recent Zephyr version.

SocIt2Em, 2020-06-03:
@pabigot understood and will do. A lot of problems have occurred. I managed to fix most of them by making class members static. I have asked platform IO to update. started regretting going down the c++ route, however will trod along.

carlescufi, 2020-06-19:
I am going to close this for now, but @SocIt2Em please feel free to reopen if this proves to be an issue on master.

closed 2020-06-19 by carlescufi