zephyrproject-rtos/zephyr #25693: ESP WiFi MPU Fault causes zephyr fatal error

akofoed, 2020-05-28:
**Describe the bug**
Using Zephyr master branch (updated 27/5-2020)
I am using the ESP32 as WiFi modem which is working quite well most of the time. However, occasionally I get this error and then Zephyr gives up on life:

```
[01:58:02.933,000] <inf> APP_MGMT: Sending status...
[01:58:32.968,000] <inf> APP_MGMT: Sending status...
[01:59:03.003,000] <inf> APP_MGMT: Sending status...
[01:59:32.038,000] <inf> APP_MGMT: Sending status...
[02:00:02.073,000] <inf> APP_MGMT: Sending status...
[02:00:32.107,000] <inf> APP_MGMT: Sending status...
[02:01:02.143,000] <inf> APP_MGMT: Sending status...
[02:01:07.179,000] <err> wifi_esp_offload: Failed to send data: link 1, ret -11
[02:01:17.181,000] <err> wifi_esp_offload: Failed to send data: link 1, ret -116
[02:01:27.183,000] <err> wifi_esp_offload: Failed to close link 1, ret -116
[02:01:27.183,000] <inf> APP_MQTT: MQTT client disconnected -11
[02:01:27.183,000] <err> os: ***** MPU FAULT *****
[02:01:27.183,000] <err> os:   Instruction Access Violation
[02:01:27.183,000] <err> os: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0x20009ac0
[02:01:27.183,000] <err> os: r3/a4:  0x20001d94 r12/ip:  0x000081d1 r14/lr:  0x0803fa23
[02:01:27.183,000] <err> os:  xpsr:  0x60000000
[02:01:27.183,000] <err> os: Faulting instruction address (r15/pc): 0x20002f48
[02:01:27.183,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[02:01:27.183,000] <err> os: Current thread: 0x20001d94 (unknown)
[02:01:27.260,000] <err> os: Halting system
```
I have seen it work for up to 8 hours but mostly it will throw the registers after anywhere between 30 minutes and some hours.

My hope is that @tsvehagen would be so kind as to look at this problem using the above error-codes that the driver is throwing out (-11 and -116). I will happily help with testing.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and application that regularly sends data using the ESP WiFi driver
2. Let it run for some hours
3. See error

**Expected behavior**
I would expect the esp_offload.c to handle the error in a manner that does not bring down the system.

**Impact**
Using a watchdog is a temporary fix.

**Screenshots or console output**
See above.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA: Master branch (27/5-2020)

**Additional context**
N/A

tsvehagen, 2020-05-28:
Could you please enable the following. It would help a bit in understanding what happens.
```
CONFIG_WIFI_LOG_LEVEL_DBG=y
CONFIG_MODEM_LOG_LEVEL_DBG=y
CONFIG_MODEM_CONTEXT_VERBOSE_DEBUG=y
```

ioannisg, 2020-05-28:
Looks like CPU tries to execute from RAM, maybe due to stack corruption.
Enable CONFIG_HW_STACK_PROTECTION, to see whether there's a stack overflow. (I assume you're building with CONFIG_ARM_MPU=y)

akofoed, 2020-05-28:
Thanks @tsvehagen for your reply. I have now enabled the 3 debug options as you requested and it is now running (producing a lot of output). I will update here in some hours when the issue returns.

@ioannisg Yes, it is a the STM32F429 nucleo board using the ESP32 as WiFi. CONFIG_ARM_MPU=y is set in the .config so I presume yes to your question. I will just let it run using @tsvehagen debug lines and then we can decide to enable the HW_STACK_PROTECTION. Is that OK?

ioannisg, 2020-05-28:
Just enable altogether HW STACK Protection, and the suggestions wrt LOG levels :)

akofoed, 2020-05-28:
@tsvehagen Got this now:
```
[00:26:32.448,000] <dbg> modem_cmd_handler: RECV
52 65 63 76 20 36 39 20  62 79 74 65 73          |Recv 69  bytes   
[00:26:32.452,000] <dbg> modem_cmd_handler: RECV
53 45 4e 44 20 4f 4b                             |SEND OK          
[00:26:32.452,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [SEND OK] (len:7)
[00:26:32.453,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045661, data 0x0
[00:26:32.453,000] <dbg> wifi_esp_offload.esp_sendto: link 1, timeout -1717986919
[00:26:32.453,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69
[00:26:32.453,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[00:26:32.456,000] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:26:32.456,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:26:32.466,000] <dbg> modem_cmd_handler: RECV
52 65 63 76 20 36 39 20  62 79 74 65 73          |Recv 69  bytes   
[00:26:32.467,000] <dbg> modem_cmd_handler: RECV
53 45 4e 44 20 4f 4b                             |SEND OK          
[00:26:32.467,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [SEND OK] (len:7)
[00:26:32.467,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045661, data 0x0
[00:26:32.467,000] <dbg> wifi_esp_offload.esp_sendto: link 1, timeout -1717986919
[00:26:32.467,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69
[00:26:32.467,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[00:26:42.469,000] <dbg> wifi_esp_offload._sock_send: Failed to send command
[00:26:42.469,000] <err> wifi_esp_offload: Failed to send data: link 1, ret -116
[00:26:42.469,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045661, data 0x0
[00:26:42.469,000] <dbg> wifi_esp_offload.esp_sendto: link 1, timeout -1717986919
[00:26:42.469,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69
[00:26:42.469,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[00:26:52.470,000] <dbg> wifi_esp_offload._sock_send: Failed to send command
[00:26:52.470,000] <err> wifi_esp_offload: Failed to send data: link 1, ret -116
[00:26:52.471,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x0, data 0x0
[00:26:52.471,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 43 4c  4f 53 45 3d 31          |AT+CIPCL OSE=1   
[00:27:02.473,000] <err> wifi_esp_offload: Failed to close link 1, ret -116
[00:27:02.473,000] <inf> APP_MQTT: MQTT client disconnected -5
[00:27:02.473,000] <err> os: ***** MPU FAULT *****
[00:27:02.473,000] <err> os:   Instruction Access Violation
[00:27:02.473,000] <err> os: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0x20009ac0
[00:27:02.473,000] <err> os: r3/a4:  0x20001d94 r12/ip:  0x00002f9a r14/lr:  0x080400cb
[00:27:02.473,000] <err> os:  xpsr:  0x60000000
[00:27:02.473,000] <err> os: Faulting instruction address (r15/pc): 0x20002f48
[00:27:02.473,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:27:02.473,000] <err> os: Current thread: 0x20001d94 (unknown)
[00:27:02.474,000] <err> os: Halting system
```

@ioannisg I will enable the stack protection and give it another go

akofoed, 2020-05-28:
@ioannisg It has been running for one hour and 15 minutes now without issues. I have been through the log and I cannot see any errors. But I do get dropped messages like this:
```
[01:08:07.153,000] <dbg> modem_cmd_handler: RECV
52 65 63 76 20 31 33 33  20 62 79 74 65 73       |Recv 133  bytes  
[01:08:07.154,000] <dbg> modem_cmd_handler: RECV
53 45 4e 44 20 4f 4b                             |SEND OK          
[01:08:07.154,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [SEND OK] (len:7)
[01:08:18.118,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045735, data 0x0
[01:08:18.118,000] <dbg> wifi_esp_offload.esp_sendto: link 1, timeout -1717986919
[01:08:18.118,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69
--- 11 messages dropped ---
[01:08:18.118,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[01:08:18.122,000] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[01:08:18.122,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
```
I have these settings in the prj.conf:

    CONFIG_WIFI_ESP_RX_STACK_SIZE=2048
    CONFIG_WIFI_ESP_WORKQ_STACK_SIZE=4096

Are you saying that I could be one of these stack-size settings that is too small?

I will let it run and update again.

Thank you very much for you help :)

ioannisg, 2020-05-28:
It can be any thread of the kernel overflowing their stack.

 (MPU faults can be related to anything, though. But since one of the most common reasons for faults of this kind is stack corruption, the first thing to do is to enable HW support to detect and report stack overflows)

tsvehagen, 2020-05-28:
During send, if timeout is non-zero (as in this case), the calling thread is actually used so it could be the calling threads stack that is too small.

akofoed, 2020-05-28:
Just got this:
```
[02:41:47.408,000] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[02:41:47.408,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[02:41:47.422,000] <dbg> modem_cmd_handler: RECV
52 65 63 76 20 31 33 33  20 62 79 74 65 73       |Recv 133  bytes  
[02:41:47.423,000] <dbg> modem_cmd_handler: RECV
53 45 4e 44 20 4f 4b                             |SEND OK          
[02:41:47.423,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [SEND OK] (len:7)
[02:41:58.349,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045735, data 0x0
[02:41:58.349,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69986919
[02:41:58.349,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[02:41:58.353,000] <dbg> modem_cmd_handler: RECV
45 52 52 4f 52                                   |ERROR            
[02:41:58.353,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [ERROR] (len:5)
[02:41:58.353,000] <dbg> wifi_esp_offload._sock_send: Failed to send command
[02:41:58.353,000] <err> wifi_esp_offload: Failed to send data: link 1, ret -5
[02:41:58.353,000] <dbg> wifi_esp_offload.esp_recv: link_id 1, timeout 0, cb 0x8045735, data 0x0
[02:41:58.353,000] <dbg> wifi_esp_offload.esp_sendto: link 1, timeout -1717986919
[02:41:58.353,000] <dbg> wifi_esp_offload._sock_send: link 1, len 69
[02:41:58.353,000] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 49 50 53 45  4e 44 3d 31 2c 36 39    |AT+CIPSE ND=1,69 
[02:41:58.356,000] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[02:41:58.378,000] <dbg> modem_cmd_handler: RECV
31 2c 43 4c 4f 53 45 44                          |1,CLOSED         
[02:41:58.378,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [1,CLOSED] (len:8)
[02:41:58.381,000] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[02:41:58.381,000] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[02:41:58.381,000] <inf> APP_MQTT: MQTT client disconnected -5
[02:41:58.381,000] <err> net_mqtt: Failed to disconnect transport!
[02:41:58.381,000] <inf> APP_MQTT: MQTT client disconnected -9
[02:41:58.381,000] <err> APP_MQTT: mqtt_publish_app_event: MQTT Error publishing message: -5
�[02:41:58.381,000] <err> os: ***** MPU FAULT *****
[02:41:58.381,000] <err> os:   Instruction Access Violation
[02:41:58.381,000] <err> os: r0/a1:  0x20003030  r1/a2:  0x20010aa8  r2/a3:  0x20001e98
[02:41:58.381,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00002e84 r14/lr:  0x08026a2b
[02:41:58.381,000] <err> os:  xpsr:  0x61000000
[02:41:58.381,000] <err> os: Faulting instruction address (r15/pc): 0xfffffffa
[02:41:58.381,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[02:41:58.381,000] <err> os: Current thread: 0x20001d94 (unknown)
[02:41:58.520,000] <err> os: Halting system
```
@tsvehagen which is the calling thread stack? RX or WORKQ? And do you have a recommended value?

tsvehagen, 2020-05-28:
> @tsvehagen which is the calling thread stack? RX or WORKQ? And do you have a recommended value?

The thread that calls send/sendto. Probably the one that does mqtt publish. Try increasing the size to something big like 4k and see what happens. It feels like it could be some bad error handling in the driver as well though. I will need to check that more closely.

akofoed, 2020-05-28:
I have been trying different things:
1) Creating a mutex that ensures MQTT only transmits or receives 
2) Playing with different stack-size settings
Both of above does not seem to have any different effect.
But I can provoke the problem if I send data too fast to Zephyr/ESP32.

This is my prj.conf for the ESP WiFi:
```
CONFIG_WIFI_ESP=y
CONFIG_WIFI_ESP_NAME="esp-wifi-modem"
CONFIG_WIFI_ESP_RX_STACK_SIZE=2048
CONFIG_WIFI_ESP_RX_THREAD_PRIORITY=3
CONFIG_WIFI_ESP_WORKQ_STACK_SIZE=4096
CONFIG_WIFI_ESP_WORKQ_THREAD_PRIORITY=3
# CONFIG_WIFI_ESP_PASSIVE_TCP=y
CONFIG_WIFI_ESP_AT_VERSION_2_0=y
```
If I enable the passive_tcp option it does not work.
I suspect that the ESP32 usart serial link is not fast enough (115200) so I tried to increase it to 2x and 4x but then the ESP WiFi says "Connection request failed". Only 115200 seems to work.
Anyway, this is how far I have gotten.
Thanks

tsvehagen, 2020-05-29:
@akofoed What is your sending threads priority?

FYI: Passive mode for ESP32 does work but you will need https://github.com/zephyrproject-rtos/zephyr/pull/25404 and a pretty new firmware on the ESP32. Espressif have made some fixes for passive mode which has not been in any release yet so you will have to build the firmware yourself.

akofoed, 2020-05-29:
I have two threads sending stuff, one on prio 6 and one that runs prio 4. So I thought that setting the ESP to 3 would be a good idea

akofoed, 2020-06-09:
I will close this for now. I have no solution for the issue.

closed 2020-06-09 by akofoed

tsvehagen, 2020-06-12:
> I will close this for now. I have no solution for the issue.

I will try to recreate and see if I can fix it when I find the time!

tsvehagen, 2020-07-24:
@akofoed FYI there has been some updates to the esp wifi driver so it might be worth to try again with the latest master.