zephyrproject-rtos/zephyr #25701: MPU FAULT in nvs test on nrf52840dk_nrf52840

markus-becker-tridonic-com, 2020-05-28:
**Describe the bug**

We are having issues with NVS on nrf52840dk_nrf52840.

The test tests/subsys/settings/functional/nvs fails with an MPU fault.

**To Reproduce**
Steps to reproduce the behavior:
`./scripts/sanitycheck --device-testing --device-serial /dev/ttyACM0 -p nrf52840dk_nrf52840 -T tests/subsys/settings/functional/nvs`

Output on UART:
```
*** Booting Zephyr OS build v2.3.0-rc1-353-g4653b4e63f88  ***
Running test suite settings_test_suite
===================================================================
starting test - test_clear_settings
PASS - test_clear_settings
===================================================================
starting test - test_support_rtn
PASS - test_support_rtn
===================================================================
starting test - test_register_and_loading
I: 8 Sectors of 4096 bytes
I: alloc wra: 0, ff0
I: data wra: 0, 0
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0xf8000
E: r0/a1:  0x00100000  r1/a2:  0x00000000  r2/a3:  0x80000000
E: r3/a4:  0x4001e000 r12/ip:  0xffffffff r14/lr:  0x0000345b
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x000034a2
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2000030c (unknown)
E: Halting system
```

**Expected behavior**
No MPU fault.

**Impact**
Unusable NVS on nrf52840dk_nrf52840?

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK,
 - Commit SHA or Version used: master 4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d

**Additional context**
NVS is used as storage backend for OpenThread.

ioannisg, 2020-05-28:
@markus-becker-tridonic-com @nvlsianpu  maybe this is because you need MPU to ALLOW FLASH WRITE, please check

nvlsianpu, 2020-05-28:
@markus-becker-tridonic-com As @ioannisg points need to use proper configuration for run it on HW boards. Will try to ad a such here.

markus-becker-tridonic-com, 2020-05-28:
yes, the prj.conf in fcb and file are setting it.

nvlsianpu, 2020-05-28:
@markus-becker-tridonic-com  here patch with proper configuration #25708

markus-becker-tridonic-com, 2020-05-28:
https://github.com/zephyrproject-rtos/zephyr/pull/25708 fixes this.

PR #25708 (2020-05-28): tests/subsys/settings/functional/nvs: add nrf52xx configuration
Added configuration which allows to run test-suite
on nrf52840dk_nrf52840 and nrf52dk_rf52832 boards.

fixes #25701
commit 48af9703 (2020-05-28): tests/subsys/settings/functional/nvs: add nrf52xx configuration

Added configuration which allows to run test-suite
on nrf52840dk_nrf52840 and nrf52dk_rf52832 boards.

fixes #25701
closed 2020-05-28 by carlescufi