zephyrproject-rtos/zephyr #25720: [Coverity CID :210073] Arguments in wrong order in drivers/modem/wncm14a2a.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/drivers/modem/wncm14a2a.c

Category: API usage errors
Function: `pkt_setup_ip_data`
Component: Drivers
CID: [210073](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210073)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nashif, 2020-07-21:
```
#if defined(CONFIG_NET_UDP)
509        if (sock->ip_proto == IPPROTO_UDP) {
   CID 210073 (#1 of 1): Arguments in wrong order (SWAPPED_ARGUMENTS)swapped_arguments: The positions of arguments in the call to net_udp_create do not match the ordering of the parameters:

    dst_port is passed to src_port
    src_port is passed to dst_port

 [show details]
510                if (net_udp_create(pkt, dst_port, src_port)) {
511                        return -1;
512                }
513
514                hdr_len += NET_UDPH_LEN;
515        } else
516#endif
```

nashif, 2020-07-21:
@jukkar  can you please take a look at this one?

jukkar, 2020-08-05:
The swapping is intentional so this is false positive.

closed 2020-08-05 by jukkar