zephyrproject-rtos/zephyr #25723: [Coverity CID :210559] Unsigned compared against 0 in subsys/net/ip/tcp2.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/subsys/net/ip/tcp2.c

Category: Control flow issues
Function: `tcp_in`
Component: Networking
CID: [210559](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210559)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2020-07-24:
This looks like a legit bug??? Every time we use tcp_data_get() function there is a test for '<0' but the return value is a size_t and can never be negative. Is this logic really wanting to compare '==0'?

dleach02, 2020-07-24:
Bumping to medium so we can get some feedback to see if it should be a low priority bug

jukkar, 2020-07-27:
The function can return -ENOBUFS value to indicate an error. We should return 0 in this case instead. Other alternative would be to change the return type to be `ssize_t` and return -1 instead. I would vote to return 0 here.

jukkar, 2020-07-27:
> Other alternative would be to change the return type to be `ssize_t` and return -1 instead. I would vote to return 0 here.

After more consideration and in order to minimize the code changes, I opted to return -1 and change the return type of the offending function.

dleach02, 2020-07-27:
I like the idea of propagating up a negative error code so the -ENOBUFS is reasonable. But not sure what the higher level code is expecting.

PR #27165 (2020-08-05): net: tcp2: Check data length properly
Return -1 in tcp_data_get() if we run out of memory. The original
return type of tcp_data_get() was unsigned and the return value <0
was not checked properly.

Fixes #25723
Coverity-CID: 210559
commit e73e9321 (2020-08-05): net: tcp2: Check data length properly

The original return type of tcp_data_get() was unsigned and the
return value <0 was not checked properly.

Fixes #25723
Coverity-CID: 210559
closed 2020-08-05 by jukkar