zephyrproject-rtos/zephyr #25731: [Coverity CID :210568] Argument cannot be negative in tests/net/socket/socketpair/src/test_socketpair_happy_path.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/net/socket/socketpair/src/test_socketpair_happy_path.c

Category: Error handling issues
Function: `happy_path`
Component: Tests
CID: [210568](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210568)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-05-28:
@cfriedt

cfriedt, 2020-05-28:
I don't have permission to view this error on coverity.com

PR #25805 (2020-06-03): net: sockets: socketpair: fix socketpair coverity issues
I have no permissions to access the actual errors on coverity.com (error 401), but please let me know if this helps.

Fixes #25793
Fixes #25788
Fixes #25780
Fixes #25737
Fixes #25731 
Fixes #25736 
Fixes #25727 
Fixes #25797
commit 15a2a6dd (2020-06-03): tests: socket: socketpair: assert recvfrom returns >= 0

Fixes #25731
Coverity-CID: 210568