zephyrproject-rtos/zephyr #25733: [Coverity CID :210575] Argument cannot be negative in tests/posix/eventfd/src/main.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/posix/eventfd/src/main.c

Category: Error handling issues
Function: `test_eventfd_read_nonblock`
Component: Tests
CID: [210575](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210575)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pfalcon, 2020-06-30:
@tsvehagen: And another one ;-).

dleach02, 2020-08-14:
This is a false positive as there is an assert test to ensure that fd >= 0

closed 2020-08-14 by dleach02