zephyrproject-rtos/zephyr #25734: [Coverity CID :210580] Argument cannot be negative in tests/posix/eventfd/src/main.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/posix/eventfd/src/main.c

Category: Error handling issues
Function: `test_eventfd_poll_timeout`
Component: Tests
CID: [210580](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210580)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pfalcon, 2020-06-30:
@tsvehagen: Can you please look into this?

dleach02, 2020-08-14:
wrong argument type for close()

```
   	7. Condition ret == 0, taking true branch.
   	8. Condition ret == 0, taking true branch.
   	9. Condition "poll ret %d", taking true branch.
   	10. Condition "poll ret %d", taking true branch.
103        zassert_true(ret == 0, "poll ret %d", ret);
104
   	
CID 210580 (#1 of 1): Argument cannot be negative (NEGATIVE_RETURNS)
11. negative_returns: fd is passed to a parameter that cannot be negative.
105        close(fd);
106}
107
```

closed 2020-08-14 by dleach02