zephyrproject-rtos/zephyr #25737: [Coverity CID :210585] Unchecked return value in samples/net/sockets/socketpair/src/socketpair_example.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/samples/net/sockets/socketpair/src/socketpair_example.c

Category: Error handling issues
Function: `zephyr_app_main`
Component: Samples
CID: [210585](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210585)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-05-28:
@cfriedt can you check this one?

cfriedt, 2020-05-28:
I can try. Unfortunately, I cannot see the actual error messages from coverity because I lack permissions on the site.

PR #25805 (2020-06-03): net: sockets: socketpair: fix socketpair coverity issues
I have no permissions to access the actual errors on coverity.com (error 401), but please let me know if this helps.

Fixes #25793
Fixes #25788
Fixes #25780
Fixes #25737
Fixes #25731 
Fixes #25736 
Fixes #25727 
Fixes #25797
commit 9cc82a03 (2020-06-03): samples: sockets: socketpair: check return of pthread_create

Fixes #25737
Coverity-CID: 210585
closed 2020-06-03 by carlescufi