zephyrproject-rtos/zephyr #25742: [Coverity CID :209943] Pointless string comparison in tests/lib/devicetree/src/main.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/lib/devicetree/src/main.c

Category: Incorrect expression
Function: `test_parent`
Component: Tests
CID: [209943](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=209943)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

mbolivar-nordic, 2020-07-07:
@MaureenHelm is there some way to whitelist the entire file? This is never going to stop otherwise:

https://github.com/zephyrproject-rtos/zephyr/issues/25987
https://github.com/zephyrproject-rtos/zephyr/issues/25985
https://github.com/zephyrproject-rtos/zephyr/issues/24041
https://github.com/zephyrproject-rtos/zephyr/issues/24040
...

The devicetree APIs are macro based, and APIs which "return" strings simply macro-expand to string literals.

The test cases check that the "returned" string matches its expected value by using `strcmp(DT_FOO(....), "expected value")`, which coverity apparently doesn't understand is a valid thing to be doing.

mbolivar-nordic, 2020-07-07:
I'm also getting 401 unauthorized from coverity when I attempt to log in with this GitHub account.

mbolivar-nordic, 2020-07-07:
marked fp, would be nice if the coverity scanner could be configured away from generating these issues in the test cases

closed 2020-07-07 by mbolivar-nordic