zephyrproject-rtos/zephyr #25747: [Coverity CID :210596] Assign instead of compare in subsys/logging/log_output_syst.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/subsys/logging/log_output_syst.c

Category: Incorrect expression
Function: ``
Component: Other
CID: [210596](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210596)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2020-08-15:
```
127static void platform_handle_init(struct mipi_syst_handle *systh)
128{
129#if defined(MIPI_SYST_PCFG_LENGTH_FIELD)
   	CID 210584: Assign instead of compare (PW.ASSIGN_WHERE_COMPARE_MEANT) [select issue]
130        MIPI_SYST_ENABLE_HANDLE_LENGTH(systh, 1);
131#endif
132
133#if defined(MIPI_SYST_PCFG_ENABLE_TIMESTAMP)
   	
CID 210596 (#1 of 1): Assign instead of compare (PW.ASSIGN_WHERE_COMPARE_MEANT)
1. assign_where_compare_meant: use of "=" where "==" may have been intended
134        MIPI_SYST_ENABLE_HANDLE_TIMESTAMP(systh, 1);
135#endif
136}
137
```

dleach02, 2020-08-15:
This is an intentional design of the underlying MIPI macro and not a bug.

closed 2020-08-15 by dleach02