zephyrproject-rtos/zephyr #25749: [Coverity CID :210033] Unintentional integer overflow in drivers/sensor/mpr/mpr.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/drivers/sensor/mpr/mpr.c

Category: Integer handling issues
Function: `mpr_convert_reg`
Component: Drivers
CID: [210033](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210033)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2020-07-21:
@sven-hm will you please take a look?

sven-hm, 2020-07-22:
@MaureenHelm of course, i'll have a look! When i follow the links to coverity.com i get an "401: Unauthorized" though. `cppcheck` does not detect the issue.
Is there another way to detect this issue locally?

MaureenHelm, 2020-07-22:
> @MaureenHelm of course, i'll have a look! When i follow the links to coverity.com i get an "401: Unauthorized" though. `cppcheck` does not detect the issue.
> Is there another way to detect this issue locally?

Thank you. You should have access now to the coverity project.

sven-hm, 2020-07-22:
> You should have access now to the coverity project.

I still get the same error. Or do i need to change some settings, too? I never used coverity before...

dleach02, 2020-07-24:
```
 91static inline void mpr_convert_reg(const uint32_t *reg, int64_t *value)
 92{
   	
CID 210033 (#1 of 1): Unintentional integer overflow (OVERFLOW_BEFORE_WIDEN)
overflow_before_widen: Potentially overflowing expression (*reg - 1677722U) * 25U with type unsigned int (32 bits, unsigned) is evaluated using 32-bit arithmetic, and then used in a context that expects an expression of type int64_t (64 bits, signed).
   	To avoid overflow, cast either *reg - 1677722U or 25U to type int64_t.
 93        *value = (*reg - MPR_OUTPUT_MIN) * (MPR_P_MAX - MPR_P_MIN);
 94        *value *= MPR_CONVERSION_FACTOR;
 95        *value /= MPR_OUTPUT_RANGE;
 96        *value += MPR_P_MIN;
 97}
 98 ```

Snippet from coverity

PR #27044 (2020-08-06): drivers: sensor: mpr: fix integer handling issue
This PR aims to fix #25749 .
Also adds a codeowner to /drivers/sensor/mpr and fixes indentation in code documentation.
closed 2020-08-06 by carlescufi