zephyrproject-rtos/zephyr #25750: [Coverity CID :210066] Unintentional integer overflow in include/sys/time_units.h

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/include/sys/time_units.h

Category: Integer handling issues
Function: `z_tmcvt`
Component: Other
CID: [210066](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210066)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

IRISZZW, 2020-05-29:
this issue will cause tests/kernel/timer/timer_api failure in emsdp_em7d_esp board.
https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/tests/kernel/timer/timer_api/src/timer_convert.c#L163-L179

When `sys_rand32_get()` return a very big value, it will cause test failure.
And this will happen every time you run in emsdp_em7d_esp board.

root cause:

https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/include/sys/time_units.h#L109-L112

serial output:

```
*** Booting Zephyr OS build v2.3.0-rc1-334-gd72a9140feab  ***
Running test suite timer_api
===================================================================
starting test - test_time_conversions

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/timer_convert.c:157: test_conversion: (diff <= maxdiff && diff >= mindiff is false)
Convert ERR from ERRhz to 0hz (= ERR) failed. diff -25000000 should be in [25000000:ERR]
FAIL - test_time_conversions
===================================================================
starting test - test_timer_duration_period
PASS - test_timer_duration_period
===================================================================
starting test - test_timer_period_0
PASS - test_timer_period_0
===================================================================
starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
starting test - test_timer_periodicity
PASS - test_timer_periodicity
===================================================================
starting test - test_timer_status_get
PASS - test_timer_status_get
===================================================================
starting test - test_timer_status_get_anytime
PASS - test_timer_status_get_anytime
===================================================================
starting test - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
starting test - test_timer_k_define
PASS - test_timer_k_define
===================================================================
starting test - test_timer_user_data
PASS - test_timer_user_data
===================================================================
starting test - test_timer_remaining
PASS - test_timer_remaining
===================================================================
starting test - test_timeout_abs
PASS - test_timeout_abs
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

@vonhust @ruuddw

pabigot, 2020-05-29:
The Coverity issue is not with line 111, but with:

https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/include/sys/time_units.h#L118

and the complaint is that the multiplu is done with 32-bit unsigned values which produces a truncated 32-bit unsigned value, but the return value from the function is 64-bit unsigned which Coverity takes as evidence that the full-precision multiplication is desired.

Absent the comment from @IRISZZW I would say this is a false positive for Coverity because the fact that `result32` is true means that all call sites are expected to cast the return value to 32-bits.  The use of 32-bit multiply here is intentional and loss of data due to overflow is expected when the user asked for only 32-bits of result.  @andyross should review this.

For the test failure described in the comment above there's a clue in:

> Convert ERR from ERRhz to 0hz (= ERR) failed. diff -25000000 should be in [25000000:ERR]

The code here is incorrect because all of the output specifiers are `%lld`, while the values passed are either `u32_t` or `u64_t`.  It's also wrong because it's using signed output for several values that are unsigned.  `ERR` seems to arise where the libc determines that the passed value was not type-correct.

Please apply the fix in #25821 to the test and report back with the corrected output including the to and from frequencies (because 0 is clearly not valid).  It's possible there's a bug, but to know that we need to now what the actual parameters were.

IRISZZW, 2020-06-01:
@pabigot In my case, the test failure is caused by line 112.
https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/include/sys/time_units.h#L112
When` sys_rand32_get() `return a very big value, such as 0xFFFFFF00(it is possible).
https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/tests/kernel/timer/timer_api/src/timer_convert.c#L176
then in line 110, t add off will overflow 32 bit, and lin112 will return zero which is unexpected.
https://github.com/zephyrproject-rtos/zephyr/blob/3a04536487797825d28160479a30ad0b72dca7cf/include/sys/time_units.h#L110

pabigot, 2020-06-01:
OK, I see what might be happening here.  Nonetheless, that isn't what this issue is about.  I've opened 
#25870 to track what you see.

pabigot, 2020-06-29:
I propose to close this as "Intentional"/minor/ignore in Coverity with the following diagnosis.  Will anybody second that decision?

The complaint is that the multiply is done with 32-bit unsigned values which produces a truncated 32-bit unsigned value, but the return value from the function is 64-bit unsigned which Coverity takes as evidence that the full-precision multiplication is desired.  However the function also verifies `result32` is set indicating the caller will explicitly truncate the result to a 32-bit unsigned value.  This behavior is well-defined and intended.

andyross, 2020-06-29:
Exactly.  It's an **intentional** overflow, with documentation.

pabigot, 2020-06-29:
Thanks; updated Coverity.

closed 2020-06-29 by pabigot