zephyrproject-rtos/zephyr #25793: [Coverity CID :210561] Resource leak in tests/net/socket/socketpair/src/test_socketpair_unsupported_calls.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/net/socket/socketpair/src/test_socketpair_unsupported_calls.c

Category: Resource leaks
Function: `test_socketpair_unsupported_calls`
Component: Tests
CID: [210561](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210561)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-05-28:
@cfriedt can you check this one?

cfriedt, 2020-06-03:
@pabigot said this can be ignored.

My comments from coverity:

> In this case, accept should only return a number other than -1 if socketpair's accept is malfunctioning. We assert that the result is -1 as expected. The resource leak may be ignored because it is test code, according to pabigot

closed 2020-06-03 by cfriedt

pabigot, 2020-06-03:
The "False Positive" assessment I would propose is:

The code confirms that accept() returned an error and did not produce a handle, so losing it is not a concern.  See issue #25793 for history.

[Actively managed](https://github.com/zephyrproject-rtos/zephyr/pull/25805#issuecomment-636013212) so leaving this to @jukkar to update Coverity.

PR #25805 (2020-06-03): net: sockets: socketpair: fix socketpair coverity issues
I have no permissions to access the actual errors on coverity.com (error 401), but please let me know if this helps.

Fixes #25793
Fixes #25788
Fixes #25780
Fixes #25737
Fixes #25731 
Fixes #25736 
Fixes #25727 
Fixes #25797