zephyrproject-rtos/zephyr #25796: [Coverity CID :210579] Uninitialized scalar variable in tests/net/socket/socketpair/src/test_socketpair_happy_path.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/net/socket/socketpair/src/test_socketpair_happy_path.c

Category: Uninitialized variables
Function: `happy_path`
Component: Tests
CID: [210579](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210579)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-05-28:
@cfriedt can you check this one?

cfriedt, 2020-05-28:
I get 401 unauthorized on the synopsis website.

commit 3f30ce21 (2020-06-03): tests: socket: socketpair: init param before passing by ref

Fixes #25796
Coverity-CID: 210579