zephyrproject-rtos/zephyr #25797: [Coverity CID :210607] Uninitialized scalar variable in tests/net/socket/socketpair/src/test_socketpair_happy_path.c

zephyrbot, 2020-05-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/4653b4e63f886a50ac7b72f8d47ba2950ab2dd0d/tests/net/socket/socketpair/src/test_socketpair_happy_path.c#L139

Category: Uninitialized variables
Function: `happy_path`
Component: Tests
CID: [210607](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210607)

Details:

```
133    
134             msghdr.msg_iov = &iovec;
135             msghdr.msg_iovlen = 1;
136             iovec.iov_base = (void *)expected_msg;
137             iovec.iov_len = expected_msg_len;
138    
>>>     CID 210607:  Uninitialized variables  (UNINIT)
>>>     Using uninitialized value "msghdr". Field "msghdr.msg_name" is uninitialized when calling "sendmsg". [Note: The source code implementation of the function has been overridden by a builtin model.]
139             res = sendmsg(sv[i], &msghdr, 0);
140    
141             zassert_not_equal(res, -1, "sendmsg(2) failed: %d", errno);
142             actual_msg_len = res;
143             zassert_equal(actual_msg_len, expected_msg_len,
144                       "did not sendmsg entire message");
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-05-28:
@cfriedt can you take a look at this one?

cfriedt, 2020-05-28:
> @cfriedt can you take a look at this one?

Yeah, can it wait a couple of hours?

PR #25805 (2020-06-03): net: sockets: socketpair: fix socketpair coverity issues
I have no permissions to access the actual errors on coverity.com (error 401), but please let me know if this helps.

Fixes #25793
Fixes #25788
Fixes #25780
Fixes #25737
Fixes #25731 
Fixes #25736 
Fixes #25727 
Fixes #25797
commit 79728a63 (2020-06-03): tests: socket: socketpair: init param before passing by ref

Fixes #25797
Coverity-CID: 210607
closed 2020-06-03 by carlescufi