zephyrproject-rtos/zephyr #25811: K22F USB Console/Shell

asob, 2020-05-29:
I've been testing out Zephyr's USB connectivity options on FRDM-K22F and got some questions:

1.  `samples/subsys/usb/console/` compiles and runs okay as long as the USB is connected. If usb is not connected the application will not run, this happens, as I can see it because of flag `CONFIG_USB_UART_DTR_WAIT=y`. My understanding was that if i compile the app with `CONFIG_USB_UART_DTR_WAIT=n` the application will run immediately, however after compiling and uploading it to target the USB CDC Serial is not being recognized by the host operating system. Is it all possible to have a console via USB without the DTR flag?
2. Is it possible to have shell redirected to USB Serial? The documentation is kinda of cryptic regarding this.

jfischer-phytec-iot, 2020-05-29:
There is a bug that prevent initialization of the usb if DTR_WAIT is not enabled, I already mentioned it on the slack but forgot to fix. @asob Please try the patch below, I will open a PR later.

```diff
diff --git a/drivers/console/uart_console.c b/drivers/console/uart_console.c
index b3c57366c3..660d0b5241 100644
--- a/drivers/console/uart_console.c
+++ b/drivers/console/uart_console.c
@@ -601,14 +601,14 @@ static int uart_console_init(struct device *arg)
 
        __ASSERT_NO_MSG(uart_console_dev);
 
-#if defined(CONFIG_USB_UART_CONSOLE) && defined(CONFIG_USB_UART_DTR_WAIT)
+#if defined(CONFIG_USB_UART_CONSOLE)
        int ret;
 
        ret = usb_enable(NULL);
        if (ret != 0) {
                return ret;
        }
-
+#if defined(CONFIG_USB_UART_DTR_WAIT)
        while (1) {
                u32_t dtr = 0U;
 
@@ -618,6 +618,7 @@ static int uart_console_init(struct device *arg)
                }
        }
        k_busy_wait(1000000);
+#endif
 #endif
 
        uart_console_hook_install();
```

asob, 2020-05-29:
@jfischer-phytec-iot 

Thanks for the patch, for some reason `git apply <patch>` didn't work fork me, kept throwing the same errors:
```
~/zephyrproject/zephyr$ git apply ~/usb_dtr_fix.patch
error: patch failed: drivers/console/uart_console.c:601
error: drivers/console/uart_console.c: patch does not apply
```

But I went ahead and just modified the `uart_console.c` manually, the results are as follows:

1. Board is powered on, code starts executing
2. Connect USB, CDC Serial is detected in Device Manager
3. Open up terminal (PuTTY) -> Console output is visible 
4. Disconnect USB while PuTTY is still running
5. Reconnect USB -> CDC Serial is detected in Device Manager
6. Open up terminal -> No console output 

If after step (3) PuTTY is closed and then USB is disconnected and reconnected, console output re-appears.

PR #25894 (2020-06-02): drivers: console: fix USB initialization
Initialize USB even if USB_UART_DTR_WAIT is not enabled.

Fixes: #25811
commit 6da7e415 (2020-06-02): drivers: console: fix USB initialization

Initialize USB even if USB_UART_DTR_WAIT is not enabled.

Fixes: #25811
closed 2020-06-02 by carlescufi