zephyrproject-rtos/zephyr #25812: Question about the Peer_Address_Type of HCI_LE_Create _Connection command of Bluetooth Host

lylezhu2014, 2020-05-29:
I encountered an issue that Zephyr V2.2.0 BLE host (named A device) cannot connect to a peer (named B device) with an RPA address.

The background is,
a.	Both of peers have performed the bonding procedures and distributed their respective IRKs. 
b.	The peer device has been added into resolving list. And resolution of Resolvable Private Addresses has be enabled.
c.	B device is advertising with RPA.
d.	The controller of A device is Marvell 88W8977. According to response of HCI_Read_Local_Supported_Commands, the controller does not support the LE Set Privacy Mode command.

After A device disconnected from B device, I tried to create a new connection by calling bt_conn_create_le on A device side. 
But the connection between A device and B device was not established by calling function bt_conn_create_le with the peer address type 0x00 (Public Device Address), or 0x02 (Public Identity Address).
I found if the peer address type parameter of bt_conn_create_le is 0x02, the parameter Peer_Address_Type of the HCI_LE_Create_Connection command will be changed to 0x00. 

My question is, why is peer address type 0x02 changed to 0x00 when creating a connection? I suppose the Peer_Address_Type of the HCI_LE_Create_Connection command should be 0x02.

And I did the following additional tests,
If the parameter Peer_Address_Type of the HCI_LE_Create_Connection command was 0x00, the connection between A device and B device was not established.
If the parameter Peer_Address_Type of the HCI_LE_Create_Connection command was 0x02, the connection between A device and B device was established.

joerchan, 2020-06-08:
@lylezhu2014 Thanks for reporting. You have encountered an ambiguous part of the LE Privacy feature that was later clarified by adding the LE Set Privacy Mode command.

We should use 0x02/0x03 as peer address type when the controller does not support the LE Set PRivacy Mode Command.

lylezhu2014, 2020-06-09:
@joerchan Do you know which release will add the feature?  Thank you!

joerchan, 2020-06-16:
> @joerchan Do you know which release will add the feature? Thank you!

@lylezhu2014 Hopefully the next one (2.4).

> And I did the following additional tests,
If the parameter Peer_Address_Type of the HCI_LE_Create_Connection command was 0x00, the connection between A device and B device was not established.
If the parameter Peer_Address_Type of the HCI_LE_Create_Connection command was 0x02, the connection between A device and B device was established.

Do you have access to a sniffer, could you attach sniffer log of when it works?
Or if you have zephyr host on both sides. Could you use `bt_conn_get_info` and print all the 4 different addresses that was used, on both sides.
Example from `subsys/bluetooth/shell/bt.c` on all 4 addresses:
```
		print_le_addr("Remote", info.le.dst);
		print_le_addr("Local", info.le.src);
		print_le_addr("Remote on-air", info.le.remote);
		print_le_addr("Local on-air", info.le.local);
```

lylezhu2014, 2020-06-18:
I cannot provide the sniffer log, but whether it is OK if I provide the HCI raw data? 
Here is HCI raw data [HCI_log_RT1060_8977_with_peer_add_type_2.txt](https://github.com/zephyrproject-rtos/zephyr/files/4798341/HCI_log_RT1060_8977_with_peer_add_type_2.txt)